espressif/arduino-esp32 #298: SimpleBLE doesn't compile

BroncoTc, 2017-04-05:
In SimpleBLE.cpp, it contains function calls to btStart() btStarted() btStop(). These functions are in esp32-hal-bt.h ,but the include tree doesn't contain this file.

mamiksik, 2017-04-09:
I have the same issue.

BroncoTc, 2017-04-09:
I guess the reason why it's not included in the platformIO installation is that something is not implemented yet. When putting everything I can find in the repo together, the linker returned an error, which is because bt.h has no .c/cpp implementation file with it. This is essential for the hal implementation.

me-no-dev, 2017-04-09:
You probably need to refresh something on PlatformIO. The files are in the repository :) just browse here and you'll see. If they are not in your PIO build, you are running a quite old version of this repo

mamiksik, 2017-04-09:
Thanks for fast response! Yep I found it here in repo, however when I compile my program it isn't there. And yes, I updated my project (and pio). Moreover I tried to create new project by using `pio --board esp32dev` and still it isn't working. Any other idea? Or did I miss something?

MrDadaGuy, 2017-04-10:
+1, same problem

me-no-dev, 2017-04-10:
Try this: http://docs.platformio.org/en/latest/platforms/espressif32.html#using-arduino-framework-with-staging-version

mamiksik, 2017-04-12:
That helped. 👍 Thanks.

closed 2017-04-16 by me-no-dev