espressif/arduino-esp32 #301: CPU frequency affects WiFi and BLE functionality

Smrtokvitek, 2017-04-06:
I have the "HIMALAYA ESP32" board and I tried the arduino-esp32 but with strange behaviour. When I set the cpu frequency to 80MHz any example from WiFi and BLE causes CPU halt. If I use 40MHz option WiFi works but the BLE still causes CPU halt.

Can it be possible HW problem (design flaw maybe)?

[image]

me-no-dev, 2017-04-06:
CPU or Flash frequency? I think you are talking about flash frequency and that depends on the chip used. Generally it should not upset the WiFi/BT stacks regardless of the setting

Smrtokvitek, 2017-04-06:
You are right, my bad I'm used to change the cpu frequency from that kind of menu for different Arduino boards. Nevertheless it acts like I've described it. So is it possible the esp module is bad?

me-no-dev, 2017-04-06:
Try giving it a better power source, use better USB cable, things like that might help. The module is probably OK, but it's badly powered

Smrtokvitek, 2017-04-06:
I've tried different cable and different power source. It is the same. It looks like I cannot make it work on 80MHz. 40MHz setting is working now.

I found out why the `ble` halts the CPU. I have to put `WiFi.mode(WIFI_STA)` before first `ble.begin` command and  then it works even if I don't want to use the wifi.

me-no-dev, 2017-04-06:
great and strange at the same time (about the flash frequency) :)

closed 2017-04-06 by me-no-dev

gggabo, 2018-06-18:
@Smrtokvitek hello, you have any example to code, please.