espressif/arduino-esp32 #302: I2S

alecapu80, 2017-04-06:
When will be available a I2S library for recording and playback via I2S bus like: 
https://www.arduino.cc/en/Reference/I2S ??

me-no-dev, 2017-04-06:
I have something similar already working but it's not ready for release.

alecapu80, 2017-04-06:
Can you send it to me to study it and try to understand how to use I2S???

filipeaguiar22, 2017-04-07:
I am also interested in the code. I would like to parse A2DC to I2S  data.
If you could share the code, it would be a great advance, for me.

alecapu80, 2017-04-10:
@me-no-dev can't you provide us your working code? We can be useful for testing, improving, etc etc

chenkaiyao, 2017-04-10:
@ me-no-dev can you release the I2S library! it is very important funtion to us

me-no-dev, 2017-04-10:
I will when it's done. Currently I have other urgent things to take care of :)
You can use IDF's API through Arduino also.
https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/i2s.h

AdySan, 2017-04-10:
Here's a good esp-idf project that uses i2s audio https://github.com/MrBuddyCasino/ESP32_MP3_Decoder

alecapu80, 2017-04-10:
Yes I know but this is a play example (output). There isn't a recording (input) example? ???

NickWaterton, 2017-06-11:
Just FYI, I have a test i2s input program working using two ics43434 i2s microphones (left and right).

alecapu80, 2017-06-12:
Have you some sample code??
Thank you

closed 2018-02-10 by alecapu80

DaTebe, 2018-03-13:
Is there now an input example?