espressif/arduino-esp32 #309: Error Compiling when used as ESP IDF component.

alwint3r, 2017-04-12:
Hi,

I got this compilation error right after updating my IDF version to the latest version on master branch.

```
arduino-esp32/cores/esp32/esp32-hal-bt.c: In function 'btStart':
arduino-esp32/cores/esp32/esp32-hal-bt.c:30:9: error: too few arguments to function 'esp_bt_co
ntroller_init'
```

I just noticed that the `esp_bt_controller_init` function signature has changed to:
`esp_err_t esp_bt_controller_init(esp_bt_controller_config_t *cfg);`

However, I managed to solve the compilation error by directly adding `NULL` as an argument to the function. I haven't noticed something weird after that, since I'm not using bluetooth (not yet).

But, I think that's not the desired solution. See [here](https://github.com/espressif/esp-idf/blob/master/components/bt/bt.c#L223).

Thanks

skateone, 2017-04-12:
i go the same now... trying your proposed fix

alwint3r, 2017-04-12:
@skateone please note that my solution won't make your bluetooth working, it's just getting rid of the compilation error.

I'm not quite sure. Would it be ok to add the `esp_bt_controller_config_t *cfg` parameter to the `btStart` function ? So we don't hardcode anything which is bad in my case because the bluetooth won't work.

me-no-dev, 2017-04-12:
I've been thinking about this and I might have found the reason. What is really strange is that builds are tested and no compilation error was thrown.

skateone, 2017-04-12:
cleared the compilation error by nullifying the arguments...
BT operation will be tomorrows problem for me

me-no-dev, 2017-04-12:
done :)

closed 2017-04-12 by me-no-dev

alwint3r, 2017-04-12:
Thanks @me-no-dev !

beyondJohn, 2017-06-14:
Nullifying esp_bt_controller_init() did not work for me. The following did work bet the init working.
`esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    esp_bt_controller_init(&bt_cfg);`
I found the above code in the following post submitted by Nicholas3388. Search the page for &bt_cfg to find the references.
[BLE indicate/notification issue #568](https://github.com/espressif/esp-idf/issues/568)