espressif/arduino-esp32 #311: PCNT not able to get counts

rdheiliger, 2017-04-13:
I am trying to use PCNT in this program. I am not getting any counts when i read the counter value. The program compiles and i get back ESP_OK for the configuration and the read counter value lines. I am using a pulse generator to input pulses on pins 27 and 14. Any thoughts?

I tried using interrupts to count pulses on a pin, but interrupts can only handle 3 to 4 pulses/second and inputs 27 and 14 work with interrupts.

void initPcnt(){

//initialize counter for flow0
pcnt_config_t pcnt_config = {
27, /!< Pulse input gpio_num, if you want to use gpio16, pulse_gpio_num = 16, a negative value will be ignored /
PCNT_PIN_NOT_USED, /!< Contol signal input gpio_num, a negative value will be ignored/
PCNT_MODE_KEEP, /!< PCNT low control mode/
PCNT_MODE_KEEP, /!< PCNT high control mode/
PCNT_COUNT_INC, /!< PCNT positive edge count mode/
PCNT_COUNT_DIS, /!< PCNT negative edge count mode/
32767, /*!< Maximum counter value /
0, /!< Minimum counter value /
PCNT_UNIT_0, /!< PCNT unit number /
PCNT_CHANNEL_0 /!< the PCNT channel */
};
if(pcnt_unit_config(&pcnt_config) == ESP_OK) //init unit
Serial.println("Config Unit_0 = ESP_OK");

pcnt_set_filter_value(PCNT_UNIT_0, 1);
pcnt_filter_enable(PCNT_UNIT_0);

pcnt_config_t pcnt_config2 = {
14, /!< Pulse input gpio_num, if you want to use gpio16, pulse_gpio_num = 16, a negative value will be ignored /
PCNT_PIN_NOT_USED, /!< Contol signal input gpio_num, a negative value will be ignored/
PCNT_MODE_KEEP, /!< PCNT low control mode/
PCNT_MODE_KEEP, /!< PCNT high control mode/
PCNT_COUNT_INC, /!< PCNT positive edge count mode/
PCNT_COUNT_DIS, /!< PCNT negative edge count mode/
32767, /*!< Maximum counter value /
0, /!< Minimum counter value /
PCNT_UNIT_1, /!< PCNT unit number /
PCNT_CHANNEL_0, /!< the PCNT channel */
};
if(pcnt_unit_config(&pcnt_config2) == ESP_OK) //init unit
Serial.println("Config Unit_1 = ESP_OK");

pcnt_set_filter_value(PCNT_UNIT_1, 1);
pcnt_filter_enable(PCNT_UNIT_1);

}

void readPcntCounter_0(){

if(pcnt_get_counter_value(PCNT_UNIT_0, &flow0Counter) == ESP_OK)
Serial.println("Read pcnt unit 0");
delay(10);
Serial.print("flow0Counts = ");
Serial.println(flow0Counter);

}

void readPcntCounter_1(){

if(pcnt_get_counter_value(PCNT_UNIT_1, &flow1Counter) == ESP_OK)
Serial.println("Read pcnt unit 1");
delay(10);
Serial.print("flow1Counts = ");
Serial.println(flow1Counter);

}

rdheiliger, 2017-04-14:
i tried another way of initializing pcnt:
this does not result in getting counts either;

  //configure for flow1counter
  
    pcnt_counter_pause(PCNT_UNIT_0);
         //set what pins to use for counter
    pcnt_set_pin(PCNT_UNIT_0, PCNT_CHANNEL_0, 27, PCNT_PIN_NOT_USED);  
    pcnt_set_pin(PCNT_UNIT_0, PCNT_CHANNEL_1, PCNT_PIN_NOT_USED, PCNT_PIN_NOT_USED);  
    delay(10);
    pcnt_set_mode(PCNT_UNIT_0, PCNT_CHANNEL_0, PCNT_COUNT_INC, PCNT_COUNT_INC, PCNT_MODE_KEEP, PCNT_MODE_KEEP);
    pcnt_set_mode(PCNT_UNIT_0, PCNT_CHANNEL_1, PCNT_COUNT_DIS, PCNT_COUNT_DIS, PCNT_MODE_KEEP, PCNT_MODE_KEEP);
    delay(10);
    pcnt_intr_disable(PCNT_UNIT_0);
    pcnt_event_disable(PCNT_UNIT_0, PCNT_EVT_L_LIM);  //PCNT watch point event: Minimum counter value
    pcnt_event_disable(PCNT_UNIT_0, PCNT_EVT_H_LIM);  // PCNT watch point event: Maximum counter value
    pcnt_event_disable(PCNT_UNIT_0, PCNT_EVT_THRES_0);  // PCNT watch point event: threshold0 value event
    pcnt_event_disable(PCNT_UNIT_0, PCNT_EVT_THRES_1);  //PCNT watch point event: threshold1 value event
    pcnt_event_disable(PCNT_UNIT_0, PCNT_EVT_ZERO);  //PCNT watch point event: counter value zero event
    pcnt_filter_disable(PCNT_UNIT_0);
    pcnt_counter_clear(PCNT_UNIT_0);
    pcnt_counter_resume(PCNT_UNIT_0);

I ran across this   https://www.esp32.com/viewtopic.php?t=1030   topic
He states that the PCNT_PLUS_CNT_RST_Un  is set to 1 by default and will prevent the counters from counting, i see nothing in    arduino-esp32/tools/sdk/include/driver/driver/pcnt.h   that will clear this bit.

me-no-dev, 2017-04-16:
Could you please file this issue to https://github.com/espressif/esp-idf :)
Arduino does not yet provide PCNT api of it's own :)

closed 2017-04-16 by me-no-dev

onpagithub, 2017-08-29:
rdheiliger: Any reaction or solutuion ?