espressif/arduino-esp32 #312: Flash filesystem example

lbernstone, 2017-04-13:
Can someone put together an example that shows how to create and use a small filesystem on the local flash similar to how SPIFFS worked on ESP8266?  FS.h is clear enough, but assumes a filesystem object is available and healthy.

me-no-dev, 2017-04-16:
SPIFFS is coming soon to idf and Arduino for esp32. You can use SD card currently to access the file system.

lbernstone, 2017-04-16:
Ok.  Thanks!

closed 2017-04-16 by lbernstone

pikolo84, 2017-05-02:
Some example already?

Beliwars, 2017-05-19:
Hello @me-no-dev,I wonder if SPIFFS is coming soon to arduino-core? We are porting a code from esp8266 to esp32, we used SPIFFS and I was wondering if we could try to use a FAT FS instead, is this already supported in arduino/core? I can't find any example showing the implementation of a Fat FS on the Flash; as @lbernstone said with FS.h we need first a File system defined.

By the way, I have ported some esp8266 arduino libraries (for the moment only the libraries, haven't changed any example yet) I have ported HTPPClient, DNSServer, HttpUpdate, WebServer and WiFiManager; I haven't port the ESP8266Ping library, ping.h header missing on esp32 core.

Thanks in advance

igrr, 2017-05-20:
@Beliwars if you have ported some libraries, feel free to submit them as PRs (one per library). At least the ones which don't depend on the FS, such as HTTPClient, DNSServer.

lonerzzz, 2017-07-14:
@Beliwars I am tracking roadmap items under the https://github.com/espressif/arduino-esp32/projects tab.   Is the ESP8266Ping port to ESP32 something on your plate or in a repo somewhere? If I so, please let me know and I will capture this.

ondabeach, 2017-07-21:
I'm holding off on a ESP32 project until SPIFFS is available as well. Any idea how long until it's released? No pressure, I really appreciate the work you guys do, but if I have a rough idea when it'll be ready it would help me make plans :)

Cheers, Steve.

lonerzzz, 2017-07-22:
@ondabeach There is an interim implementation (https://github.com/copercini/arduino-esp32-SPIFFS). The only part currently missing is the file system upload tool.

ondabeach, 2017-07-22:
@lonerzzz you sir are a legend :)
Many thanks to you and all who contributed to the project.

skateone, 2017-07-22:
awesome lib. implemented it and it works pretty well.

ondabeach, 2017-07-22:
@skateone good to hear. I only need to store a dozen or so config variables.

skateone, 2017-07-23:
i use the NVS libs to store config data. 
Arduino has wrapped this using the Preferences.h lib... its slightly easier than working with files

ondabeach, 2017-07-23:
@skateone I use SPIFFS in my other projects so by using SPIFFS in my esp32 project I can re-use a heap of code :)

skateone, 2017-07-23:
LOL that was also my though when using the ESP32.. boy was i wrong. i have managed to get a combination of esp32 arduino and the native IDF combined to make my stuff function efficiently so far. when i say so far i am still tweeking

ondabeach, 2017-07-23:
Sounds like I'm in for some fun ;)

everslick, 2017-07-24:
Has anybody combined webserver_tng and SPIFFS yet? spiffs works, but i cannot stream a file via http like this.

```
File file = SPIFFS.open("/foo.txt", FILE_READ);
webserver.streamFile(file, "text/plain");
file.close();
```

copercini, 2017-07-24:
@everslick Did you tried the SD example (modified to SPIFFS) ?

everslick, 2017-07-24:
>
> @everslick <https://github.com/everslick> Did you tried the SD example
> (modified to SPIFFS) ?
>
yes the example sketch works flawlessly.

or do you mean the webserver filebrowser example? that one i did not try
yet.

I have to note, that on ESP8266 filenames did not have to start with '/',
but now they do not work without an leading '/'. but this is unrelated to
the problem with the webserver.

bbx10, 2017-07-25:
WebServer_tng updated.

The FSBrowser example works for sufficiently relaxed definitions of "works". Throughput is slow and sometimes the files do not completely transfer. Could still be something wrong with streamFile. Or maybe my AP because other devices are wonky right now.

Use curl to upload html, javascript, etc.

`$ curl -X POST "data=@index.htm" http://<ESP32 IP addr>/edit`

SDWebServer works about the same.

copercini, 2017-07-25:
@bbx10 any steps to reproduce it on FSBrowser ?
I tryied some html pages and this "large" [image](https://cdn.sparkfun.com/assets/home_page_posts/2/0/1/7/esp32-wroom-03-2.png) many times (cleaning browser cache every time) with zero fails

This image takes from 3 to 9 seconds to load (most times in 4 or 5 secs), I don't have a ESP8266 right now to benchmark it

everslick, 2017-07-25:
will try it a little later - hang on...

bbx10, 2017-07-25:
I power cycled my AP which has made file transfers more reliable. But still slow compared to ESP8266.

ESP8266 using git `b695898 Fixed unused-function warning` 1.8 seconds to download 512 KB file using FSBrowser.

```
$ wget -O /dev/null http://192.168.1.129/512KB.DAT
--2017-07-25 11:25:38--  http://192.168.1.129/512KB.DAT
Connecting to 192.168.1.129:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 524288 (512K) [text/plain]
Saving to: ‘/dev/null’

100%[======================================>] 524,288      288KB/s   in 1.8s   

2017-07-25 11:25:40 (288 KB/s) - ‘/dev/null’ saved [524288/524288]
```

Next is the result for ESP32. The times vary a lot. Sometimes below 10 seconds, sometimes over 30s but never as fast as the ESP8266. Both connect to the same AP.
```
$ wget -O /dev/null http://192.168.1.99/512KB.DAT
--2017-07-25 11:29:28--  http://192.168.1.99/512KB.DAT
Connecting to 192.168.1.99:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 524288 (512K) [text/plain]
Saving to: ‘/dev/null’

100%[======================================>] 524,288     23.5KB/s   in 21s    

2017-07-25 11:29:49 (24.5 KB/s) - ‘/dev/null’ saved [524288/524288]
```

everslick, 2017-07-25:
Interesting is for me, that file uploads do work, if slow. i get 8KB/s.
Downloads do not work at all. The server closes the connection after 2
seconds and the browser reports a content length mismatch. filesize is
irrelevant for that to happen. something is fishy.

bbx10, 2017-07-25:
Be sure to update all WebServer_tng files. git pull or git clone will update all files. Sounds like you are missing some fixes.

everslick, 2017-07-25:
>
> Be sure to update all WebServer_tng files. git pull or git clone will
> update all files. Sounds like you are missing some fixes.
>

ahh correct, thanx! WebServer.h was indeed not the latest. due to
nameclashes and such i have to manually rename and copy files, argl.
nevermind...

further tests show, that uploads are still dead slow (7,5KB/s), but
downlods now work great. i see 800KB/s here. but WLAN is pretty stable and
clean here.

bbx10, 2017-07-26:
My very slow downloads are only on an Adafruit Feather Huzzah32 board. I switched to an Espressif ESP32 dev board. Downloads are much faster but not as fast as an ESP8266. 

Download 512 KB file (524288 bytes) downloaded 7 times
Range: 2.0 to 7.9 seconds
Average: 5.1 seconds, 102,512 bytes/sec, 820,115 bits/sec

pbecchi, 2017-07-29:
#