espressif/arduino-esp32 #313: error make ESP-IDF and arduino

igrr, 2017-04-14:
_From @sfranchin63 on April 14, 2017 11:58_

Hi,
I follow the step indicate in : 
https://github.com/espressif/arduino-esp32
but the compiling stopped with error. I attached the error.
The code is:

#include "Arduino.h"
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "driver/gpio.h"
// test led
#include "WS2812.h"

#define LED1 0   // R_IO0
#define LED2 2   // R_IO2
#define LED_BLU 4   // R_IO4

void hello_task(void *pvParameter)
{
	theaterChaseRainbow (20);
    gpio_pad_select_gpio(LED1);
    gpio_pad_select_gpio(LED2);
    gpio_pad_select_gpio(LED_BLU);
    /* Set the GPIO as a push/pull output */
    gpio_set_direction(LED1, GPIO_MODE_OUTPUT);
    gpio_set_direction(LED2, GPIO_MODE_OUTPUT);
    gpio_set_direction(LED_BLU, GPIO_MODE_OUTPUT);
    gpio_set_level(LED1, 0);
    gpio_set_level(LED2, 0);
    gpio_set_level(LED_BLU, 0);
	while (1) {
    for (int i = 10; i >= 0; i--) {
        gpio_set_level(LED_BLU, 1);
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        gpio_set_level(LED_BLU, 0);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
/*    printf("Restarting now.\n");
    fflush(stdout);
   esp_restart();
*/
	}
}

void extern "C" void app_main()
{
	initArduino();
	hornbillLights_begin((rmt_channel_t) 0, (gpio_num_t)16, (uint16_t)150);
    xTaskCreate(&hello_task, "hello_task", 2048, NULL, 5, NULL);
}
[image]

This is what I am using :
 -a uname
MSYS_NT-6.1-WOW Stefanof-PC 2.5.2 (0.297 / 5/3) 2016-07-15 09:04 i686 Msys
$ rendere -v
GNU Make 4.2.1
$ Xtensa-ESP32-elf-cc - versione
Xtensa-ESP32-elf-cc.exe (CrossTool-NG 8d95cad) 4.8.5
$ Python -V
Python 2.7.11

Can somuone help me?
Thanks you in advance
Stefano

_Copied from original issue: espressif/esp-idf#524_

igrr, 2017-04-14:
base64 library has a bug on Xtensa, because it assumes that `char` is a signed type. There was a PR in the ESP8266 Arduino repository which fixed this, maybe should be applied here as well.

sfranchin63, 2017-04-15:
Hi @igrr 
thanks for your instructions that have allowed me to solve the warnings of cdecode.c library.
But it continues to be there the following error:

### C:/msys32/home/Stefanof/esp-idf/examples/get-started/hello_world/components/arduino/cores/esp32/esp32-hal-misc.c: In function 'initArduino':
C:/msys32/home/Stefanof/esp-idf/examples/get-started/hello_world/components/arduino/cores/esp32/esp32-hal-misc.c:84:15: error: 'ESP_ERR_NVS_NO_FREE_PAGES' undeclared (first use in this function)
     if(err == ESP_ERR_NVS_NO_FREE_PAGES){
               ^
C:/msys32/home/Stefanof/esp-idf/examples/get-started/hello_world/components/arduino/cores/esp32/esp32-hal-misc.c:84:15: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [/home/Stefanof/esp-new/esp-idf-v2.0-rc1/make/component_wrapper.mk:177: cores/esp32/esp32-hal-misc.o] **Error 1**
make: *** [/home/Stefanof/esp-new/esp-idf-v2.0-rc1/make/project.mk:382: arduino-build] **Error 2**

Stefano

[image]

igrr, 2017-04-15:
`ESP_ERR_NVS_NO_FREE_PAGES` isn't defined in the IDF version you are compiling against. Suggest updating IDF to a more recent master version.

sfranchin63, 2017-04-16:
Hi @igrr,
I followed your suggest updating IDF to a more recent version. To do this I used the version of all-in-one toolchain & MSYS zip file:
https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20170330.zip
I also installed the ESP-IDF version 2.0. But now I have another problem, the compiling stopped because  the Python is not found. Any suggestions ?

igrr, 2017-04-16:
Could you please check if this suggestion helps?

https://github.com/espressif/esp-idf/issues/493#issuecomment-291686676

sfranchin63, 2017-04-18:
Hello @igrr,
thank you for the suggestion, I found where wrong:
in the new version I must open an MSY2 terminal by running c: \ mys32 \ mingw32.exe.
But 
[image]
appeared a new message that is not a problem because the compiling running:
"It is not a valid identifier export:" dashless
ESP32-hal-misc.c: 84: 15: error: 'ESP_ERR_NVS_NO_FREE_PAGES' undeclared
 Any suggestions ?
[image]

me-no-dev, 2017-04-18:
@sfranchin63 while 2.0 is the latest released version, github actually has more changes and this is one of them. I currently sync Arduino with github IDF and should maybe start adding releases together with IDF, so things like that do not happen from now on (or at least there will be a way to compile against older code).
I sugest you replace your IDF with the github one.

closed 2017-05-06 by me-no-dev