espressif/arduino-esp32 #316: pinMode(x, INPUT_PULLUP) does not pull up on GPIO34 ... GPIO39

hcs-svn, 2017-04-16:
Tested it for all GPIOs that can be a DI. Works for all of them but not for GPIO34 ... GPIO39.
They work as input but get not pulled up. Same with INPUT_PULLDOWN.

igrr, 2017-04-16:
These input-only pins don't have PU/PD circuits.

hcs-svn, 2017-04-16:
OK, thanks for the information.
Then it's time for me to start soldering ...

closed 2017-04-16 by hcs-svn

BarryAndrews, 2018-12-11:
Wouldn't this be something worth putting in the manual?
Bit annoying to only find this out by searching once the pin fails to work as expected.

beegee-tokyo, 2018-12-11:
Good idea. Will put it into my [ESP32 WiKi](https://desire.giesecke.tk/index.php/2018/01/30/esp32-wiki-entries/)

beegee-tokyo, 2018-12-11:
Ups, already there: [Reserved GPIO’S](https://desire.giesecke.tk/index.php/2018/07/06/reserved-gpios/)

mws-rmain, 2019-12-23:
I've always found it anoying that this issue is not CLEAR from the start of development. WHY are these pins even CALLED 'GPIO's, then later found to be 'INPUT ONLY'? In my opinion, these pins should have been labelled 'GPI', since 'GPIO' actually MEANS 'General Purpose Input/OUTPUT'. If the documentation lists most pins as 'GPIO#', then lists four pins as 'GPI#', it'c clear to the designer that these pins are different, WITHOUT digging through document details.

ahmed-asaleh, 2020-06-28:
Hello, 
As I understand, all of the esp32 GPIO pins have internal pull-up & pull-down resistors, except GPIO 34 tell GPIO 39.
Is that correct? If yes, then how can I use these internal pull-up/down resistors? maybe like this (not sure) : 
`pinMode(pinNumber, INPUT_PULLUP);` or `pinMode(pinNumber, INPUT_PULLDOWN);`
Thanks in advance ;)