espressif/arduino-esp32 #319: SD library

ArtemChernatsky, 2017-04-19:
Hi all,

I have problems with mounting sd card:
```
[E][sd_diskio.cpp:708] sdcard_mount(): f_mount failed 0x(1)
Card Mount Failed
```
```
Connection:
MOSI 23
MISO 19
SCK 18
SS 5
```

[Micro SD Card Module](http://i.ebayimg.com/images/g/xY0AAOSw0fhXiEqx/s-l1600.jpg)
With [this](https://github.com/adafruit/arduino-esp32/tree/d602169ebcd126c309009e87c1e455f2c892d08f) library it works.

me-no-dev, 2017-04-19:
How are you starting the card? ```SD.begin(5);``` is how you define the SS pin

ArtemChernatsky, 2017-04-19:
I'm using `SD.begin()`, but I've recently tried:
```
SD.begin(5)
SD.begin(4)
SD.begin(26)
SD.begin(27)
```
Error is the same.
I'm using [this](http://img1.banggood.com/thumb/view/oaupload/banggood/images/E0/B2/db908e09-77c8-40de-9baa-101beae89815.JPG) board.

me-no-dev, 2017-04-19:
This is rather strange... I use the same pins and the same lib daily. Could you try to start the SD the following way?
```cpp
SD.begin(5, SPI, 8000000);
```

me-no-dev, 2017-04-19:
Also if you can change the default debug level of Arduino (option in board menu if you select dev module)

ArtemChernatsky, 2017-04-19:
With `SD.begin(5, SPI, 1000000);` error is the same. 
I'm using PlatformIO IDE how can i change debug level here?

me-no-dev, 2017-04-19:
PIO..... are you running the stage version of ESP32?
http://docs.platformio.org/page/platforms/espressif32.html#using-arduino-framework-with-staging-version

ArtemChernatsky, 2017-04-19:
Yes, ESP32 version is stage.

me-no-dev, 2017-04-19:
There should be a way for you to have the following flag while building PIO project
```-DCORE_DEBUG_LEVEL=5```

me-no-dev, 2017-04-19:
Also you can try updating or reinstalling the framework for PIO, it has proven not to update sometimes

ArtemChernatsky, 2017-04-19:
Generally build flags are adding in platformio.ini like this: `build_flags =-DCORE_DEBUG_LEVEL=5`, but I can't see difference in output. I have reinstalled PIO last thursday. 
I'll test my board with Arduino IDE later to figure out if it works correctly with it.

ArtemChernatsky, 2017-04-20:
I've tested in Arduino IDE with Core debug level "debug" for combinations:
```
SD.begin(); // Connected to pin 5
SD.begin(5); // Connected to pin 5
SD.begin(5, SPI, 1000000);  // Connected to pin 5
SD.begin(26); // Connected to pin 26
SD.begin(26, SPI, 1000000);  // Connected to pin 26
```
For all variants i recieving this error:
`[E][sd_diskio.cpp:724] sdcard_mount(): f_mount failed 0x(1)`

igrr, 2017-04-20:
It seems like the card is either not formatted, or FATFS library can not recognize the format (which will happen if you try to use an exFAT formatted card).

ArtemChernatsky, 2017-04-20:
@igrr it is formatted as FAT and I've just tested with FAT32, it also doesn't work, but both file systems work with [adafruit/arduino-esp32](https://github.com/adafruit/arduino-esp32/tree/d602169ebcd126c309009e87c1e455f2c892d08f).

me-no-dev, 2017-04-20:
BTW that Adafruit repository is really old. Lots have changed in SPI since then. Do you call SPI before or after SD init?

ArtemChernatsky, 2017-04-20:
I've tried to call `SPI.begin();` before `SD.begin(5);` but result was the same. And I've copied only SD library from Adafruit, everything else was from this repository.

me-no-dev, 2017-04-20:
Then that give even less clues :( I went through the code and given that you say you tried different debug levels and this is the only line that you get, I can't find a reason. To not get other debug means that card init went fine (detects card type), but mount failed from then on.

copercini, 2017-04-20:
Do you have anything else connected to the ESP?
Do you tried with another SD card?

ArtemChernatsky, 2017-04-21:
@copercini only micro SD card module was connected. I've tested with other cards and it works with 16Gb and 512Mb cards, so it seems the problem was in that card.

ArtemChernatsky, 2017-04-21:
Is there any limitations for file names? I couldn't read file /config.json but when I renamed it to /config.txt it was read successfully.

me-no-dev, 2017-04-21:
could you list your SD card and see what it gives?
```cpp
void listDir(const char * dirname){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = SD.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}
```

ArtemChernatsky, 2017-04-21:
It gives in PIO:
```
Listing directory: /
  DIR : /SYSTEM~1
  FILE: /CONFIG~1.JSO  SIZE: 450
  FILE: /CONFIG.TXT  SIZE: 450
```
In Arduino IDE it gives:
```
Listing directory: /
  DIR : /System Volume Information
  FILE: /config.json  SIZE: 450
  FILE: /config.txt  SIZE: 450
```
And reading is working fine here.
It seems that errors are only with PIO, thanks.

closed 2017-04-21 by ArtemChernatsky

me-no-dev, 2017-04-21:
old PIO version of esp32-arduino

copercini, 2017-04-21:
A long time ago I was the same dilemma, adafruid versus me-no-dev Sd library.... So it's my empirical experience it both:

Adafruid's mount and start even if the card or the hardware has a problem, but it will crash in runtime at any time

Me-no-dev lib don't start if everything is OK with the hardware and I never see it crashes in runtime.

robin-lockwood, 2017-12-08:
I am also having this problem with my ESP32 Thing that is on a breadboard.  The fix for me was to solder my sd breakout board to my ESP32 Thing on a PCB prototyping board.  My pinouts are the default pinouts for SPI the code is the same, the boards are practically from the same batches.

CS:5, SCK:18, MOSI:19, MISO:23

Maybe there is some signal degradation when using a breadboard?  Haven't experienced a mounting error since I switched over.

me-no-dev, 2017-12-11:
could be caused by less than perfect connection or too long wires that experience cross-talk at high SPI frequencies.

Develectronic, 2017-12-21:
I had also troubles with esp32 and the sd library. I connected a ssd1351 coler oled display and an microSD card connector (as described in the sd card library description).  I tried to run the code from the adafruit SSD1351 Display library but I got the error:

[E][sd_diskio.cpp:708] sdcard_mount(): f_mount failed 0x(1)
Card Mount Failed

I looked really hard for the error, but can't find anything. Then I tried once a microSD card (without HC) instead of several microSDHC card and it worked like a charm. I need to check other microSD cards with FAT16 filesystem. But I assume that there is a issue in the library, that it accepts FAT32 not correctly.

cr13, 2018-05-08:
hello,
I don't know what to try anymore, I get the "Card Mount Failed" ruling.
"I'm using the pins (CS: 5, SCK: 18, MOSI: 19, MISO: 23), Arduino IDE compiler and I'm programming the SD_test library and I don't have to test it anymore. 
I have tried two 512MB and 8GB microSDs in FAT16 and FAT32 formats and always the same result.
What could I prove or what am I doing wrong?

stickbreaker, 2018-05-08:
@cr13 Try adding pullups to MISO and MOSI, I use 3.3k to the 3.3v Vdd pin.

Chuck.

cr13, 2018-05-08:
@stickbreaker I don`t undertand that you want say with "Try adding pullups to MISO and MOSI", I use 3,3v in pin VCC, I try with resistor 3.3k

cr13, 2018-05-10:
@stickbreaker
ok, I try and I comment.
Thank you very much.

cr13, 2018-05-11:
hello @stickbreaker,
I'm still making the same mistake.
Attached is a picture of my circuit.
https://drive.google.com/file/d/1dCF0EeIApVoeN3FsdAdVMd1DU_NuqwsW/view?usp=sharing

Is it possible that the module is wrong?

pipi61, 2018-05-11:
Hi!
check your sd adapter pin voltage level....
The esp use 3.3V level. This sd card use 5Vin-output/3.3V sdcard level converter, check this how to work 3.3/3.3V

Sivapraveen-Entuple, 2021-08-30:
im working on the sdcard with same file but i need to use different pins for MISO, MOSI, CLK and CS as well, how can i configure the pins for sdcard module via SPI

chegewara, 2021-08-30:
https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/examples/SPI_Multiple_Buses/SPI_Multiple_Buses.ino

https://github.com/espressif/arduino-esp32/blob/master/libraries/SD/src/SD.h#L31