espressif/arduino-esp32 #320: For using timers, OTA doesn't work smooth

andybrk, 2017-04-20:
If I define a timer like:

hw_timer_t *     timer = NULL;
void StartTimer(void) {
  timer = timerBegin(0, 80, true); 
  timerAttachInterrupt(timer, &timer100msec, true);
  timerAlarmWrite(timer, 100000, true);
  timerAlarmEnable(timer);
  Serial.println("Timer started");
}

void StopTimer(void) {
  timerDetachInterrupt(timer);
  timerStop(timer);
  timerAlarmDisable(timer);
  timerEnd(timer);
  timer = NULL; 
  Serial.println("Timer stopped");
}

and start the timer in the SW then the OTA upgrade works very stuttery and breaks often (no response). I have to repeat more times for successful OTA programming. Serial flashing is no problem. First I try the timer stop with the command  "timerEnd(timer)" only and add one by one command. Nothing did help.

This doesn't help, too:

  ArduinoOTA.onStart([]() {
    String type;
    StopTimer();
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    Serial.println("Start updating " + type);
  });

andybrk, 2017-04-22:
I made a short complete program to reproduce the problem. With this the OTA will be crashed.

`#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "xxxxx";
const char* password = "xxxxxx";

hw_timer_t *     timer = NULL;
int32_t tcount;

void timer100ms(void);

void StartTimer(void) {
  timer = timerBegin(0, 80, true); // TimerID=0, Prescaler=80, Countup=true
  // Attach timer100msec function to our timer.
  timerAttachInterrupt(timer, &timer100ms, true);
  // Set alarm to call onTimer function every second (value in microseconds). Repeat the alarm (third parameter)
  timerAlarmWrite(timer, 100000, true);
  // Start an alarm
  timerAlarmEnable(timer);
  Serial.println("Timer started");
}

void StopTimer(void) {
  timerEnd(timer);
  timer = NULL; 
  Serial.println("Timer stopped");
}

void timer100ms(void) {
  Serial.println(tcount++);
}

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.print("*");
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  StartTimer();
}

void loop() {
  ArduinoOTA.handle();
}`

The output during OTA:
235
236
Start updating sketch
**237
238
239
240
241
242
243
244
245
246
247
248
249
250
251
252
253
254
255
256
257
258
259
260
261
262
263
264
****Guru Meditation Error of type IllegalInstruction occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400d0e62  PS      : 0x00060031  A0      : 0x40081b39  A1      : 0x3ffc0be0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x000000ff  A5      : 0x4008559c  
A6      : 0x3ffcdec0  A7      : 0x3ff42000  A8      : 0x80080bec  A9      : 0xe00a1400  
A10     : 0x3ffc1094  A11     : 0x3ffc1094  A12     : 0x20000000  A13     : 0x00000001  
A14     : 0x00000400  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d0e62:0x3ffc0be0 0x40081b39:0x3ffc0c00

CPU halted.

I'm very surprised that the SW runs continued during OTA.

igrr, 2017-04-22:
@me-no-dev this will be resolved once you can switch to esp_intr_alloc instead of hardcoded interrupt numbers. Interrupt allocator will disable non-IRAM interrupts for you while the cache is disabled.

@andybrk as a temporary workaround declare your timer interrupt handler with IRAM_ATTR attribute and don't use functions such as Serial in the interrupt handler.

andybrk, 2017-04-22:
Thanks, this did help for the crash, but not for the broken communication during OTA upgrade. I did remove too much. ;-)
Now I declare the irq handler with the IRAM_ATTR attribute and add

portENTER_CRITICAL_ISR(&timerMux);
.....
portEXIT_CRITICAL_ISR(&timerMux);

like the timer example. 
Additional I stop the timer during OTA upgrade in this sequence:

ArduinoOTA.onStart( {
String type;
    StopTimer();
......

The update with this small SW is smooth now. But my real SW has still the same problem (the update is stuttering and the connection is broken often). 

It' possible that this have to do with the size of the *.bin? The sketch shows 499237 bytes. 
It's possible to extend the flash for the program if use it 1MB only? Because the SPIFS is not working and don't need the flash for that. I think the Wroom32 should have 4MB.

me-no-dev, 2017-04-22:
@igrr will do :)
@andybrk 

> It's possible to extend the flash for the program if use it 1MB only?

I'm not sure I understand your question. There is 1MB partition for the running firmware and another 1MB for OTA. That means that you can have a sketch up to 1MB that you can upload.

andybrk, 2017-04-22:
I though the ESP32 has the same standard configuration like the ESPP8266. Here I configure normally a 1MB and a 3MB partition.
If I have the full 1MB for the sketch and 1MB for OTA then the size should not be the problem.
Q: Where is another 2MB? The Wroom32 should have 4MB

me-no-dev, 2017-04-22:
ESP32 has partition scheme that defines what goes where and how big it is. Arduino uses this configuration: https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default.csv

andybrk, 2017-04-22:
Thanks!
I will try to find out continued the reason for the broken OTA update. For the moment I have no idea...

andybrk, 2017-04-23:
Now I found the reason: During download has a drastic increasing of the signal strength of the ESP. I check it the the router, too. It was often a bad connection with high fluctuation. 
The reason was an electrolyte capacitator to much close to the antenna. ;-)

closed 2017-04-23 by andybrk