espressif/arduino-esp32 #323: esp_deep_sleep_get_wakeup_cause() not returning ext1 wakeup cause

PRosenb, 2017-04-21:
In the following example code, the PIN 15 is configured to wake the chip up on HIGH level.
It works well when PIN 15 is not connected. In this case, the chip wakes up immediately after going to sleep and detects which PIN caused the wake up.

It does not work though when PIN 15 is connected to ground when the chip goes to sleep. When PIN 15 is disconnected from ground, the chip wakes up, but the wake up cause is ESP_DEEP_SLEEP_WAKEUP_UNDEFINED.

```c++
#include <esp_deep_sleep.h>
#define WAKEUP_INT_PIN 15

void setup() {
  esp_deep_sleep_wakeup_cause_t wakeupCause = esp_deep_sleep_get_wakeup_cause();

  Serial.begin(115200);
  pinMode(WAKEUP_INT_PIN, INPUT_PULLUP);

  Serial.print("wakeupCause: ");
  switch (wakeupCause) {
    case ESP_DEEP_SLEEP_WAKEUP_EXT1:
      Serial.println("ESP_DEEP_SLEEP_WAKEUP_EXT1");
      break;
    case  ESP_DEEP_SLEEP_WAKEUP_UNDEFINED:
      Serial.println("ESP_DEEP_SLEEP_WAKEUP_UNDEFINED");
      break;
    default:
      Serial.println("unknown");
  }

  esp_err_t pdConfigError = esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  if (pdConfigError != ESP_OK) {
    Serial.println("pdConfigError");
  }

  uint64_t mask = 0;
  mask |= 1 << WAKEUP_INT_PIN;
  esp_err_t enableWakeupError = esp_deep_sleep_enable_ext1_wakeup(mask, ESP_EXT1_WAKEUP_ANY_HIGH);
  if (enableWakeupError != ESP_OK) {
    Serial.println("enableWakeupError");
  }
}

void loop() {
  // sleep 5 seconds after startup
  if (millis() > 5000) {
    Serial.println("sleep");
    delay(100);
    esp_deep_sleep_start();
  }
}
```

The console output is as follows:

```bash
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:2016
load:0x40078000,len:7780
ho 0 tail 12 room 4
load:0x40080000,len:252
entry 0x40080034
wakeupCause: ESP_DEEP_SLEEP_WAKEUP_UNDEFINED
sleep
```
Disconnect PIN 15 from ground:
```bash
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:2016
load:0x40078000,len:7780
ho 0 tail 12 room 4
load:0x40080000,len:252
entry 0x40080034
wakeupCause: ESP_DEEP_SLEEP_WAKEUP_UNDEFINED
```

igrr, 2017-04-21:
@me-no-dev i think you may need to rebuild IDF libs with "Extra delay in deep sleep wake stub" option set to 500.

me-no-dev, 2017-04-21:
@igrr the current version of the libs has it defined to 1000. Isn't that OK?
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/config/sdkconfig.h#L78

andriyadi, 2017-04-26:
I experienced the same problem. When I set "Extra delay in deep sleep wake stub" to 2000, it seems working. Is that the solution? What's the implication?

Thanks.

me-no-dev, 2017-04-26:
Are you saying that 1000 was not enough?

andriyadi, 2017-04-26:
1000 is not enough on my board*.
I jump straight to 2000, not 1100, not 1500, though. And it seems working well so far.

* using my custom board based on ESP-WROOM-32.

PRosenb, 2017-04-30:
Many thanks for analyzing this case.
It works well now. @me-no-dev, have you done a change?

me-no-dev, 2017-05-06:
OK I have changed it to 2000 in the latest commit :) https://github.com/espressif/arduino-esp32/commit/376961d16844a3190bfc409e74c6acd0a4c93224

I'm closing the issue now. Feel free to reopen if that did not fix it for all.

closed 2017-05-06 by me-no-dev