espressif/arduino-esp32 #328: Arduino IDE: there are no ESP32-Boards to slelect.

kendo55, 2017-04-23:
i followed the steps to install Arduino ESP32 support on Windows.
Everythings works as described.
But in the Arduino IDE under select your board in Tools > Board menu.--> there are no ESP32-Boards to slelect.
Thank for your help in advance.
Kendo

me-no-dev, 2017-04-28:
you must have installed the core in the wrong folder. Please do give more info than just "there are no ESP32-Boards to select". Obviously this is not a problem for 99.99% of the people, so it must be something on your end, but with the scarce info that you give, nobody can determine the issue or help you.

kendo55, 2017-04-29:
Sorry! You are right!
With the described procedure, 
... install the latest Arduino IDE Windows Installer ...
With the emphasis on "Windows Installer"
And a new Installation,
it works fine!!!
My proplem was, that i tried to install without the Windows Installer.
I took the "Windows ZIP file" and expanded it. 
Like i had done, with many other Arduino IDE's.
I created a "portable" directory and cloned there the
https://github.com/espressif/arduino-esp32.git
.
My question was: how i can tell the Adrduino  IDE,
that it have to work, with the ESP32 expansion toogether?
Thanks Kendo

me-no-dev, 2017-04-29:
if you use the zipped variant of arduino, you have to use install esp32 into the hardware subfolder of the arduino unzipped folder.
[unzipped-arduino]/hardware/espressif/esp32...

closed 2017-04-30 by me-no-dev

rajeevwhiz, 2018-04-20:
I seem to have the same problem.. I am on Mac. I created /Arduino/libraries/hardware/espressif/ and `git clone https://github.com/espressif/arduino-esp32.git esp32` there. 

Ran get.py in tools 

> $ ./get.py
> System: Darwin, Info: Darwin-17.5.0-x86_64-i386-64bit
> Platform: x86_64-apple-darwin
> Tool xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz already downloaded
> Extracting xtensa-esp32-elf-osx-1.22.0-80-g6c4433a-5.2.0.tar.gz
> Tool esptool-da31d9d-macos.tar.gz already downloaded
> Extracting esptool-da31d9d-macos.tar.gz
> Tool mkspiffs-0.2.2-arduino-esp32-osx.tar.gz already downloaded
> Extracting mkspiffs-0.2.2-arduino-esp32-osx.tar.gz
> Renaming mkspiffs-0.2.2-arduino-esp32-osx to mkspiffs
> Done

I restarted Arduino, but I don't see ESP32 board under

DF5FD, 2018-08-14:
works still a year later - great!!!!