espressif/arduino-esp32 #329: peek() in Stream class problem

miomir1981, 2017-04-24:
Hi all,

I have found that peek() method is not implemented properlly in Stream class. It returns 0. 

I discovered this when I was testing ThingSpeak functionality. Response code is obtained by trying to parseInt() within response form server and this is where application get stuck.

me-no-dev, 2017-05-06:
Ahhh took me a while to track it down ;) You mean that it was not implemented right on the WiFiClient class.

closed 2017-05-06 by me-no-dev

miomir1981, 2017-05-06:
Now peek() will return -1. This didn't solve the issue.

I ran into this problem when I was testing ThingSpeak for ESP32. WriteFields() call sends the data to ThingSpeak server and data is successfully published, but there is an error when it tries to parse response because it is unable to read integer code of the response. ThingSpeak tries to parse integer from response received, starting from the position in the stream where it finds "HTTP/1.1".  If peek() method returns -1, then parseInt()->peekNextDigit()->timedPeek()->peek() calls in Stream class will always return 0, i.e. result is interpreted as timeout. WiFiClient is not able to properly execute the parseInt() method. 

peek() method for ESP8266 have implementation in ClientContext class. It have access to receive buffer and it returns byte from current position of the buffer.

me-no-dev, 2017-05-07:
Yes, but on ESP32 things are different and we use the POSIX API to access the network. I'm not sure that we can peek at the data with POSIX.

miomir1981, 2017-05-07:
I think this will work. I'm getting correct answer from parseInt() method now. 

`int WiFiClient::peek()
{
    if(!available()) {
        return -1;
    }	
    uint8_t data = 0;
    int res = recv(fd(), &data, 1, MSG_PEEK);
    if(res < 0 && errno != EWOULDBLOCK) {
        log_e("%d", errno);
        stop();
    }
    return data;
}`

I found this description on [http://pubs.opengroup.org/onlinepubs/009695399/functions/recv.html](url)
It states this:

"MSG_PEEK
Peeks at an incoming message. The data is treated as unread and the next recv() or similar function shall still return this data."

miomir1981, 2017-05-17:
@me-no-dev 
Hi, did you rewiev the proposed code for peek() method or it is not correct.

Thanks

me-no-dev, 2017-05-17:
yes, sorry I'm otherwise preoccupied with work. If you have tested this and it works, please do submit a PR. It will be much easier for me to merge it :)

me-no-dev, 2017-05-22:
implemented!