espressif/arduino-esp32 #330: WifiClient setTimeout problem

der-Lemmi, 2017-04-25:
wen setting a timeout:
`Wificlient client = server.avalible();`
`client.setTimeout(2000);`
`...`
`while(client.connected()) {`
` `  `  if(client.availible()) {`
`...`
this will have no effect.

If a client is connected and sends no data -> server hangs in the client.connected() loop because the timeout will not terminate the client.

bbx10, 2017-05-16:
#377 might be related to this problem.

me-no-dev, 2017-05-22:
Can you please try the latest code?

der-Lemmi, 2017-05-25:
seems to work now!

I logged for 24h and all clients with no data requests are got the timeout :-)

closed 2017-05-25 by der-Lemmi

marioelectron, 2021-04-27:
serverClients[i].setTimeout(1);
 serverClients[i].print(PString) ;  ///  when the wireless is lost it gets stuck in the mission  serverClients[i].print(PString)  esp32 not 
                                                                                                                                                                                            //timeout