espressif/arduino-esp32 #331: No CAN bus information?

Fyod, 2017-04-25:
Hey guys,
Will CAN be implemented into Arduino for ESP32 any time soon? The info is very scarce.
Thanks!

S2K-ger, 2017-05-16:
Agreed! CAN would be really helpful! It would be great to have at least rough time perspective to plan accordingly...

copercini, 2017-09-03:
Here is a nice tutorial by @nhatuan84 about CAN and arduino ESP32: http://www.iotsharing.com/2017/09/how-to-use-arduino-esp32-can-interface.html

everslick, 2018-04-19:
No activity in this thread since nearly a year. Issue is not a bug, but a limitation, that will eventually be addressed. -> closing.

closed 2018-04-19 by everslick

Civilduino, 2019-08-06:
Hi, any plans to implement the CAN driver soon? Maybe this library is a good start https://github.com/miwagner/ESP32-Arduino-CAN ?

atanisoft, 2019-08-06:
@Civilduino with a library like that there is no reason to include a HAL version in the Arduino-esp32 code base. There are a lot of other libraries that are very similar to the one you linked and the IDF API for CAN does work very well in the Arduino side.