espressif/arduino-esp32 #332: WiFi does not name a type

Buffalchill, 2017-04-25:
Hi what am i doing wrong ? 

i am getting this error message : 

Arduino: 1.8.1 (Windows 8.1), Board: "ESP32 Dev Module, 80MHz, 921600, None"

LiOr_ESP32_17_04_25:32: error: 'WiFi' does not name a type

 WiFi.mode(WIFI_AP_STA);

 ^

LiOr_ESP32_17_04_25:33: error: 'Serial' does not name a type

 Serial.println("** Starting AP");

 ^

LiOr_ESP32_17_04_25:34: error: 'WiFi' does not name a type

 WiFi.softAP(ssid, password);

 ^

LiOr_ESP32_17_04_25:101: error: 'int index' redeclared as different kind of symbol

 int index;                  // zum abspeichern und laden von Programmen/szenen

     ^

In file included from C:\Users\Stefan Schmidt\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/stdio.h:29:0,

                 from C:\Users\Stefan Schmidt\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:27,

                 from sketch\LiOr_ESP32_17_04_25.ino.cpp:1:

C:\Users\Stefan Schmidt\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/string.h:54:15: note: previous declaration 'char* index(const char*, int)'

 char  *_EXFUN(index,(const char *, int));

               ^

C:\Users\Stefan Schmidt\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib/_ansi.h:65:30: note: in definition of macro '_EXFUN'

 #define _EXFUN(name, proto)  name proto

                              ^

C:\Users\Stefan Schmidt\Documents\Arduino\hardware\espressif\esp32\libraries\LiOr_ESP32_17_04_25\LiOr_ESP32_17_04_25.ino: In function 'void In_Data_Sequenzen_speichern()':

LiOr_ESP32_17_04_25:489: error: assignment of function 'char* index(const char*, int)'

       index = 4;

             ^

LiOr_ESP32_17_04_25:489: error: cannot convert 'int' to 'char*(const char*, int)' in assignment

LiOr_ESP32_17_04_25:496: error: lvalue required as increment operand

       index++;

            ^

LiOr_ESP32_17_04_25:505: error: invalid types 'char [65][char*(const char*, int)]' for array subscript

         if (inData[index] != '_'){

                         ^

LiOr_ESP32_17_04_25:506: error: invalid types 'char [65][char*(const char*, int)]' for array subscript

             inChar[inCharIndex] = inData[index]; 

                                               ^

LiOr_ESP32_17_04_25:507: error: lvalue required as increment operand

             index++;

                  ^

LiOr_ESP32_17_04_25:511: error: invalid types 'char [65][char*(const char*, int)]' for array subscript

         if (inData[index] == '>'){

                         ^

LiOr_ESP32_17_04_25:517: error: invalid types 'char [65][char*(const char*, int)]' for array subscript

         if (inData[index] == '_'){

                         ^

LiOr_ESP32_17_04_25:522: error: lvalue required as increment operand

         index++; 

              ^

C:\Users\Stefan Schmidt\Documents\Arduino\hardware\espressif\esp32\libraries\LiOr_ESP32_17_04_25\LiOr_ESP32_17_04_25.ino: In function 'void In_Data_Szenen_speichern()':

LiOr_ESP32_17_04_25:545: error: assignment of function 'char* index(const char*, int)'

       index = 4;

             ^

LiOr_ESP32_17_04_25:545: error: cannot convert 'int' to 'char*(const char*, int)' in assignment

LiOr_ESP32_17_04_25:552: error: lvalue required as increment operand

       index++;

            ^

LiOr_ESP32_17_04_25:561: error: invalid types 'char [65][char*(const char*, int)]' for array subscript

         if (inData[index] != '_'){

                         ^

LiOr_ESP32_17_04_25:562: error: invalid types 'char [65][char*(const char*, int)]' for array subscript

         inChar[inCharIndex] = inData[index]; 

                                           ^

LiOr_ESP32_17_04_25:563: error: lvalue required as increment operand

         index++;

              ^

LiOr_ESP32_17_04_25:567: error: invalid types 'char [65][char*(const char*, int)]' for array subscript

         if (inData[index] == '>'){

                         ^

LiOr_ESP32_17_04_25:573: error: invalid types 'char [65][char*(const char*, int)]' for array subscript

         if (inData[index] == '_'){

                         ^

LiOr_ESP32_17_04_25:578: error: lvalue required as increment operand

         index++; 

              ^

Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: C:\Users\Stefan Schmidt\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi
 Nicht benutzt: C:\Program Files\Arduino\libraries\WiFi
exit status 1
'WiFi' does not name a type

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

Buffalchill, 2017-04-25:
1. I changed my "index" variable into "indx" .... 
2. I put the WiFi configruation into the Setup routine....  

it compiled !

closed 2017-04-26 by Buffalchill

rajatksud, 2020-04-26:
that does the job - setup routine

Beat-Pixel, 2020-07-27:
I am having the same problem, maybe there's another way to fix this?

lbernstone, 2020-07-27:
If your issue is not the same as the OP (reserved named variable), perhaps you should open your own issue instead of hijacking.