espressif/arduino-esp32 #333: A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

croisez, 2017-04-26:
NB: I introduced same issue report at espressif/esptool, because I don't know actually where the problem could be: https://github.com/espressif/esptool/issues/198

Hello ESP32 lovers,
I am facing a problem with a WEMS ESP32 dev board which I bough on Aliexpress very recently:
https://fr.aliexpress.com/item/ESP32-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board/32801337284.html

I am able to flash it using Eclipse/esp-idf/xtensa-esp32-elf triplet. The /dev/tty.SLAB_USBtoUART device configured by "make menuconfig" makes the flashing process a success. Further on, my user application (one of the examples of esp-idf) starts and does what expected.

But, with Arduino IDE, this is not the same story. I have installed the toolchain as written at ESP32-Arduino github page, then selected "ESP32 dev module" as the board in Arduino IDE, and also selected /dev/tty.SLAB_USBtoUART as the flashing device.
Like I did for the Eclipse case, I reset dev board with RST and BOOT buttons. I verified to get the "DOWNLOAD" invite on the terminal, then closed the terminal pushed on flash button of Arduino IDE. All I get in Arduino logs is the following:

Sketch uses 118,890 bytes (11%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 9,492 bytes (3%) of dynamic memory, leaving 285,420 bytes for local variables. Maximum is 294,912 bytes.
/Users/lmc/Dropbox/arduino/hardware/espressif/esp32/tools/esptool --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_freq 80m --flash_mode dio --flash_size 4MB 0x1000 /Users/lmc/Dropbox/arduino/hardware/espressif/esp32/tools/sdk/bin/bootloader.bin 0x8000 /var/folders/zx/4rs36y2j0315wny99n1j7fj80000gn/T/arduino_build_538507/Blink.ino.partitions.bin 0xe000 /Users/lmc/Dropbox/arduino/hardware/espressif/esp32/tools/partitions/boot_app0.bin 0x10000 /var/folders/zx/4rs36y2j0315wny99n1j7fj80000gn/T/arduino_build_538507/Blink.ino.bin 

esptool.py v2.0-beta
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____
 
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
The programming does not detect the dev board in download mode, and the flashing does not start.
As a workaround, I hooked up an external FTDI adapter cable directly to the ESP32, and then Arduino IDE is able to start the flashing. But after, the Arduino sketch does not seems to run. My simple Blink sketch is not running (it toggles a GPIO and prints some text on Serial console).

Issue 1: I saw a post about the problem of the FreeRTOS scheduler not started on Core1, or not enabled Core1, which could resolve the Arduino sketch not running (but I don't know how to configure it within ESP32-Arduino libraries/board from inside Arduino IDE. Could it be a efuse which should be reset to enable Core1 to start?

Issue 2: Concerning my flashing issue, I don't understand why my dev module can be successfully programmed by using the Makefile from within Eclipse/esp-idf, while it ends-up with a timeout from within ArduinoIDE/ESP32-Arduino toolchain.

Anyone could help me about my two issues?
Thank you so much.

me-no-dev, 2017-04-26:
on Issue 1: Arduino is configured with both cores already when built, so nothing else is required.
on Issue 2: Could you try to lower the flashing speed? You seem to have selected 921600 and I have seen that to fail on my mac too. Try with 115200 as a safe net and report how that goes. To note, I have had success with many baud rates with FTDI chips, but not the same with others. Silabs seems to be better than WCH, but not quite as good as FTDI.

croisez, 2017-04-27:
Hello,
just tried with lowest speed (115200) without luck.

<PRE>
Sketch uses 104,282 bytes (9%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 9,492 bytes (3%) of dynamic memory, leaving 285,420 bytes for local variables. Maximum is 294,912 bytes.
/Users/lmc/Dropbox/arduino/hardware/espressif/esp32/tools/esptool --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_freq 80m --flash_mode dio --flash_size 4MB 0x1000 /Users/lmc/Dropbox/arduino/hardware/espressif/esp32/tools/sdk/bin/bootloader.bin 0x8000 /var/folders/zx/4rs36y2j0315wny99n1j7fj80000gn/T/arduino_build_145081/Blink.ino.partitions.bin 0xe000 /Users/lmc/Dropbox/arduino/hardware/espressif/esp32/tools/partitions/boot_app0.bin 0x10000 /var/folders/zx/4rs36y2j0315wny99n1j7fj80000gn/T/arduino_build_145081/Blink.ino.bin 
esptool.py v2.0-beta2
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
</PRE>

Moreover, just before flashing, I verified I got my ESP32 put into Download mode with the Arduino terminal, and I got this message:

<PRE>
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
</PRE>

I then tried flashing, without success, and after, I went back to the terminal and saw some characters arrived after the "waiting for download" string:

<PRE>
�  UU    ��  UU    ��  UU    ��  UU    ��  UU    ��  UU    ��  UU    ��  UU    ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      ��      �
</PRE>

Could it be possible that Arduino IDE does not actually close the serial port well, so that esptool is not able to see the other end?

croisez, 2017-04-27:
Another info: inside Arduino terminal, I am able to send commands and receive info from the minimalist BASIC terminal which is activated if you pull up GPIO12 while asserting RST.
This means, at least, that my communication through the SLAB chip is bidirectional and arrive well to the ESP32.

me-no-dev, 2017-04-27:
FYI arduino and idf both use the same upload script. Arduino has it in both python and binary form (to coop with missing python, modules and so on), so your issues are really, really strange. I can not really come up with why it happens...

croisez, 2017-04-27:
I own another ESP32 module, the WROOM32 module alone, which I have put on a experimentation PCB. This one, I have been able to program it through the same Arduino IDE environment.
So, I doubt that my problem is related to a bug in ESP32-Arduino toolchain.
I am a bit lost, I can't determine the problem, and I am looking for some ideas...

croisez, 2017-04-27:
When hooking another Serial adapter (FTDI) on TX/RX pins of the WEMS module, I can program it using ESP32-Arduino toolchain:

<PRE>
Sketch uses 104,282 bytes (9%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 9,492 bytes (3%) of dynamic memory, leaving 285,420 bytes for local variables. Maximum is 294,912 bytes.
/Users/lmc/Dropbox/arduino/hardware/espressif/esp32/tools/esptool --chip esp32 --port /dev/cu.usbserial-FTXQKKR2 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_freq 80m --flash_mode dio --flash_size 4MB 0x1000 /Users/lmc/Dropbox/arduino/hardware/espressif/esp32/tools/sdk/bin/bootloader.bin 0x8000 /var/folders/zx/4rs36y2j0315wny99n1j7fj80000gn/T/arduino_build_658130/Blink.ino.partitions.bin 0xe000 /Users/lmc/Dropbox/arduino/hardware/espressif/esp32/tools/partitions/boot_app0.bin 0x10000 /var/folders/zx/4rs36y2j0315wny99n1j7fj80000gn/T/arduino_build_658130/Blink.ino.bin 
esptool.py v2.0-beta2
Connecting.......
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Flash params set to 0x022f
Compressed 12352 bytes to 7462...

Writing at 0x00001000... (100 %)
Wrote 12352 bytes (7462 compressed) at 0x00001000 in 0.7 seconds (effective 147.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 105...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (105 compressed) at 0x00008000 in 0.0 seconds (effective 1534.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4108.8 kbit/s)...
Hash of data verified.
Compressed 188768 bytes to 57684...

Writing at 0x00010000... (25 %)
Writing at 0x00014000... (50 %)
Writing at 0x00018000... (75 %)
Writing at 0x0001c000... (100 %)
Wrote 188768 bytes (57684 compressed) at 0x00010000 in 5.1 seconds (effective 294.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
</PRE>

But then upon RST, the blink sketch get stuck during the boot process:

<PRE>
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:2036
load:0x40078000,len:9988
load:0x40080000,len:252
entry 0x40080034
</PRE>

Is there some binary to flash into the ESP32 module itself to make it working with the Arduino toolchain?

croisez, 2017-05-02:
Hi,
concerning the sketch upload, I tried the following:
pushing on BOOT button during flashing process (but without acting on RST button at all) allows Arduino to flash the WEMS board successfully.
There is still a problem with the sketch which does not start. I will close this issue, and also, my apologize for putting two issues into one ticket. This is perturbating.
So let us close this one.

architmuchhal12, 2017-05-23:
I am facing the same issue with the **WEMOS LOLIM 32 v1.0.0**. It does not has a BOOT button though, only the RESET button. What should I do?
_(https://forum.wemos.cc/topic/334/lolin32-v1-0-0-is-release)_

lrmoreno007, 2017-05-23:
Please try installing new CP2102 drivers http://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

With the old drivers I had many problems previously: I could not upload without pressing boot, interference with other USB devices, etc. Curiously, those old drivers worked well with all my ESP8266 boards.

Regards.

architmuchhal12, 2017-05-23:
Hey, thanks for the suggestion.
I updated the drivers and not this is what the error is - 

`Sketch uses 108830 bytes (10%) of program storage space. Maximum is 1044464 bytes.
Global variables use 9616 bytes (3%) of dynamic memory, leaving 285296 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.0-beta3
Connecting....
Uploading stub...

A fatal error occurred: Response doesn't match request
A fatal error occurred: Response doesn't match request
`
[image]

lrmoreno007, 2017-05-24:
You should not have to press Boot button, but while you solve the problem you can put a button, as in this image:
[image]

kp372, 2017-05-31:
I am getting the same issue as the topic of this issue. I tried everything mentioned above but no luck. Do anyone have any other suggestion? 
One more thing I like to mention is, I read "OK!" on the serial monitor when GPIO0 is pulled low. I don't if this is some kind of hardware status ping.

fededim, 2017-06-17:
I experienced a similar issue on a Geekcreit board and I managed to get it working by lowering the Arduino GUI's upload speed to 115200 (menu Tools -> Upload speed -> 115200, flash button does not need to be pressed). The flashing and serial output work fine but however I am still getting garbled bootloader messages on serial as if there are some serial issues.

**If this does not work try to bypass the FTDI chip, e.g. by connecting directly an USB to TTL cable (4 wires VCC, TX, RX, GND) to ESP-32 (VCC5, RX, TX, any GND, select Esp-32 Dev Module to flash in arduino gui).** In this way my board works also at full flashing speed and I did not receive anymore garbled messages from the bootloader , so I am quite sure that there is an issue with the ftdi chip on my board (near it there are some empty pads which I think should be populated with some capacitors).

Genotronex, 2017-08-01:
Press and hold the boot button untill you have progress bar in the buttom , this problem caused by the old CP2102 Driver 

i had the same issue and i solved it by press the boot button "hard reset"

G6EJD, 2017-08-11:
Have you tried turning the cpu speed down from 80 to 40Mhz when programming?

ouaibsky, 2017-08-22:
Hi

I got the same pb with this board: https://fr.aliexpress.com/item/ESP32-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board/32802136026.html 

and last VCP driver https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

trying to decrease cpu speed, upload speed without any success.

I still stuck, not being able to upload sketch :-(

Chris

kp372, 2017-08-22:
Try to monitor GPIO0 and GPIO2 while sketch is being uploaded. GPIO0 & GPIO2 both has to be pulled down which is a requirement on ESP32 to run it into boot mode unlike ESP8266.

ouaibsky, 2017-08-23:
Hi
Sorry in advance my knowledge regarding electronic is low.

I measure between GPIO2 ans GPIO and o 3.3V in any cases (whatever I reset or push the boot button, or trying to upload a sketch

Error in arduino console is 
```
esptool.py v2.1-beta1
Connecting........_
Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content
```

Then into serial monitor I got (looping)
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK

```

G6EJD, 2017-08-23:
You have to keep the boot button pressed whilst programming?

ouaibsky, 2017-08-23:
I test with pressing the BOOT button with same issue.
When pushing BOOT I can see that GPIO0 is LOW (.6v)

ouaibsky, 2017-08-23:
runnig esptool on command line give me:
```
./esptool.py --port /dev/cu.SLAB_USBtoUART flash_id
esptool.py v2.1-beta1
Connecting........__
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 0)
Uploading stub...
Running stub...
Stub running...
Manufacturer: ff
Device: ffff
Detected flash size: Unknown
Hard resetting...
``` 

There is no Device/Manufacturer, could it be an issue ?

kp372, 2017-08-23:
> I measure between GPIO2 ans GPIO and o 3.3V in any cases (whatever I reset or push the boot button, or trying to upload a sketch

Did you check GPIO2 while pressing BOOT button ? 
Pull GPIO2 to ground right from the module. Make sure you are not sorting power supply to ground.

ouaibsky, 2017-08-23:
GPIO2 remains LOW (0.0V) and GPIO0 move from 3.33V to ~0.5V

me-no-dev, 2017-08-24:
GPIO12 sets the flash power ;) make sure it's low when booting/flashing

copercini, 2017-08-24:
@ouaibsky I have exactly this board, and works here without set or press anything....

Try just change your USB cable (I got some flash problems with a cheap cable)

ouaibsky, 2017-08-24:
I tried 2, they're good quality I think, never had issue before, can flash esp8266 without any pb.
Are you under Mac OS ?

copercini, 2017-08-24:
no, just tested it on Windows 10 and 7

ouaibsky, 2017-08-24:
I'll try to get a PC with windows 10

Dennissee, 2017-09-13:
Any further information in this topic? Im still getting the same error

G6EJD, 2017-09-13:
This is how I flash the ESP32 module. 
1. Check USB port of the connection to the ESP32, mine is usually USB-12
2. Begin compilation / upload
3. Press and hold GPIO0 button - don't release
4. Press and hold GPIO2 button momentarily, the ESP32 responds with entering Flash Mode

Are you getting boot messages from the ESP32 before you start the Flash process?

Do you get messages from the ESp32 during the Flash initialisation?  

Are you monitoring the Serial port during flashing?

Have you considered the ESP32 might be fake or faulty?

Dennissee, 2017-09-13:
I have to say I'm as newbie as possible with microcontrollers as I'm pretty sure Ive got a totally different problem (I dont even understand most of your questions). I plugged in my Esp32 via USB, installed arduino-core, and drivers linked above. But looks like my PC never recognized my board. No popup or any entry in device manager. (nothing changes with boot and en button pressing)
I see two ways: 1, Im doing something (or everything :D) wrong.
2, My NodeMCU-ESP-32S is faulty

me-no-dev, 2017-09-13:
You probably need a driver ;)

Dennissee, 2017-09-13:
I installed the newest VCP driver, did you mean something else?

G6EJD, 2017-09-13:
Sometimes Windows won't recognise a USB device, so when you plug in the ESP32, (the on-board UART) is not being recognised. So what sort of ESP32 do you have, can you post a link  to the supplier / image of it please.

Dennissee, 2017-09-13:
Nevermind, Its working now!
I reinstalled driver, changed the cable, unplugged every other usb device, plugged the board. Aaand it was working.
Anyway, here is the link for the board (its hungarian): https://www.hestore.hu/prod_10037961.html

G6EJD, 2017-09-13:
Great new, yes that's a nice standard development board, nice and stable. Happy coding...

No need to press EN and REST on that board the serial port (UART) will do it for you.

Dennissee, 2017-09-13:
Thank you, and for all the great advice as well.

mauricio34, 2017-09-17:
I also have the AliExpress-ESP32 and the same ...._____....._____....._____....._____....._____-Problem when
I try to connect any serial device (i tried many different) on the RX0 (Pin GPIO03). 
When I connect to RX2 (Pin GPIO16) my devices work.

G6EJD, 2017-09-17:
rx0 and tx0 are used by the on-board UART to communicate with the ESP32 by your IDE through the microUSB, so you cannot use them unless you plan to programme the device and then disconnect it and connect another device via the microUSB or use it standalone from the IDE.

The errors you are seeing are correct as there is a conflict, so you have to use Tx/Rx 2 or SoftwareSerial to define and use any two other pins.

w8an, 2017-09-18:
Removing the ESP32 from the breadboard during programming really does make a difference. I'm using the Doit ESP32 Devkit V1 with the Arduino IDE set on board: ESP32 Dev Module with the Upload Speed baudrate at 115200. If fails with the "A fatal error occurred: Timed out waiting for packet header" when breadboarded, but succeeds when laying on the bench unconnected except for the usb cable. No button presses are required prior to upload.

me-no-dev, 2017-09-18:
that probably means that one of the bootstrap pins is floating, so when you put it on the breadboard in induces enough to pull the pin in the wrong direction (like reading ADC on floating pin) which causes you not to be able to flash it.

vahegan, 2017-10-03:
I am having this ........_____....._____....._____..... issue on a board which was uploading just fine half an hour ago. Don't know what happened. Besides, when I disconnect the board from USB or press reset on the board, sometimes my mac restarts... When I connect another similar board to the same cable and USB port, it works just fine...

G6EJD, 2017-10-03:
What board type do you have selected and what mode DIO or QIO the speed can also make a difference try at 40MHz

vahegan, 2017-10-04:
The board type selected is WEMOS LOLIN, of course I tried to downgrade to 115200 upload speed and 40MHz flash speed, but no lock. What do you mean by DIO/QIO mode? That same board was loading under these same settings, and just stopped doing so. But the board works and there is output on the serial monitor by the previous sketch (I2C scanner) that was loaded into it.

vahegan, 2017-10-04:
It seems I have found the solution to this issue on some other board. Will post it here, maybe it will be useful to someone else. 

When you click on "upload sketch" in arduino IDE, you see the sktech being recompiled , relinked and then uploaded (before the upload phase), hold the RESET button on the board, then push and hold the second button. When arduino IDE starts the upload stage (Connecting.......), release the RESET button and then in a short while release the second button.

After this procedure the board was successfully uploaded and resumed its normal functions, as before. You need to do this only once, to get the board out of the mode in which it was stuck, then it works as normal, without having to push any buttons during the upload.

G6EJD, 2017-10-04:
There's a bi-stable circuit on most development boards that does the same as pressing boot/enable for you, sounds like it got into the wrong state and going through your process rest it.

YvedD, 2017-10-08:
Hi all, 

I had the same issue as described in the topic header.

I could fix it by pushing the IO0 button while uploading the code to the "esp32doit-devkit-v1"
my question : will i always need to use that button, or is there some sort of setting that i can tweak to make the board to be flashed without pushing that button?

I'm on Platformio (totaly newbee on that !)
Totally new to FreeRTOS too :-)

Grtz

Engebot-Technology, 2017-10-21:
VIDEO THAT CLEARS, WITH HIM I STILL RESOLVE THIS PROBLEM

https://www.youtube.com/watch?v=J10J_7Ap6Oo

CONNECTING THE GPIO0 AND EN TO A 10K RESISTOR CONNECTING THE 3.3V POWER SUPPLY

1 - PUSH ON GIO0 - PUSH BUTTON - PLAYING EARTH FOR THE PINE AND SECURING
2 - PUSH ON RESET - PUSH BUTTON - PLAYING EARTH FOR THE PINE AND SECURING
3 - PUSH OFF RESET - PUSH BUTTON
4 - PUSH OFF  GPIO0 - PUSH BUTTON
5 - COMPILING THE CODE IN THE ARDUINO IDE
6 - APOS COMPILE PUSH ON THE RESET  BUTTON PLAYING EARTH FOR THE PINE

ENJOY !!!
ENGEBOT TECHNOLOGY

G6EJD, 2017-11-14:
What power source are you using, the ESP32 takes quite a lot of current at start-up, if your powering from a USB-Serial converter that is most likely your problem. You could add a 10uF capacitor across the supply pins of the ESP32 board, or larger if you have one. 

Flashing speed is 115,200.

Sequence is GPIO0 low (usually with one of the typical push buttons, then reset, then release reset and GPIO. 

Do you have pull-up resistors on your carrier board for GPIO0 and 2.

G6EJD, 2017-11-14:
OK, I thought you were using an ESP32 bro g under the ESP32 topic, please post a link to your board so we can see what type it is.

me-no-dev, 2017-11-16:
yup 8266 problem in the esp32 thread :)

closed 2017-11-16 by me-no-dev

shofixti244, 2017-11-22:
My apologies for posting in a closed issue, but having read through this issue multiple times, and exhausted the next 3 pages of google search results, I can safely say that this is certainly not a closed issue for me.

Like the original poster, i am encountering an error upon building to my ESP32 device. It looks like so:

`37.5 s
platformio run --target upload
|   |-- <SPI> v1.0
|   |-- <FS> v1.0
|-- <SPI> v1.0
Compiling .pioenvs/esp-wrover-kit/src/main.ino.o
Looking for upload port...
Auto-detected: /dev/cu.usbserial-14100
Uploading .pioenvs/esp-wrover-kit/firmware.bin
esptool.py v2.1-beta1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____
 
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2`

I am using a brand new ESP-WROVER-KIT V3, and I have tried the following:

1. Holding the boot button as I build
2. Holding the boot button as I build, then releasing it shortly in.
3. Turning the device off, holding the boot button, turning the device on, then flashing
4. Unplugging the device, turning it off, holding the boot butting, turning the device back on, plugging it in, then flashing,
5. Doing the same 4 steps with the EN button
6. Doing the same 4 steps with the EN AND Boot Buttons, and releasing each at different stages in an exhaustive attempt to make things work.
7. Using a totally different computer to build from
8. Using both ESP IDF and the ESP Arduino framework in PlatformIO
9. Grounding GPIO2 prior to, and during flashing
10. Building both plugged into a breadboard, and sitting on top of a static free surface
11. Using both platformIO and the command line to attempt the build
12. Using bare minimal empty projects.
13. Using known code that have worked for other people.
14. Praying to any god who would listen

All to no avail.
I should note, that the very first time I built to the device everything worked fine. Then all subsequent attempts have resulted in the above error.

I'm willing to just chop it up to a defective device, unless somebody else has any input to add here. That would be much appreciated. I miss sleep.

G6EJD, 2017-11-22:
Try selecting the ? FireBeetle board type and speed 40Mhz if I recall that’s how I got programming Woking on my WRover board then I recall but not exactly that the mode needs to be DIO - no need to press any buttons if v3 board

shofixti244, 2017-11-22:
@G6EJD You sir, are a godsend. Programming using the Firebeetle board works succesfully!

I still want to leverage the increased RAM of the WROVER, so should I make a clone of the Firebeetle board and artificially set it's reported RAM higher? Or would that actually matter at all in the first place? Also is there a danger in using an incorrect board profile?

Eitherway, thank you so much. This has certainly helped.

G6EJD, 2017-11-22:
I copied the FireBeetle board and named it WRover, apart from flash size I’m not aware of any compiler usage of the memory definitions so left them as they are. You can access the on-board flash but it’s not that big. Glad you have it working now.

G6EJD, 2017-11-23:
also you will need the Espressif display driver as the v3 board has a revised driver chip, it’s on github in their repository.

Shreyashvk, 2017-12-14:
Problem solved , because of manufacturing defect some esp32 chip connecter are not properly Sholder to motherboard, reheat and apply shoulder again ,and the error doesn’t appear and the sketch uploads .

shofixti244, 2017-12-14:
@Shreyashvk I am curious, as despite my above comment we do still have this issue... Could you elaborate on what you soldered exactly? I'm assuming you're talking about the ESP32 module itself and not the esp32 chip within the module?

w8an, 2018-01-01:
I simply pull down the TX0 and RX0 pins (GPIO1 & GPIO3) with a 10k resistor to ground. No more problem uploading with the ESP32 on a breadboard.

Dannsei, 2018-01-18:
**Another solution:** I got this error (and various others) with ESP32 dev board with mini USB connected to PC (taking power from there). The issue was resolved by uploading to the board with only the USB cable attached. I guess sensors, leds etc. draw current making supply unstable. It now works nicely with 921600 connection with no need to press buttons etc.

[133 more replies]