espressif/arduino-esp32 #334: Update boards.txt for Onehorse's ESP32 Dev Module

mjs513, 2017-04-26:
I picked up a nice dev board from tindie created by Kris Winer, https://www.tindie.com/products/onehorse/esp32-development-board/ .  I was wondering if you could add this board to the boards.txt.  The only difference is that I changed DIO to DOUT and made a name change but here are the changes I made:

```
esp32t.name=ESP32 Dev Module Onehorse

esp32t.upload.tool=esptool
esp32t.upload.maximum_size=1044464
esp32t.upload.maximum_data_size=294912
esp32t.upload.wait_for_upload_port=true

esp32t.serial.disableDTR=true
esp32t.serial.disableRTS=true

esp32t.build.mcu=esp32
esp32t.build.core=esp32
esp32t.build.variant=esp32
esp32t.build.board=ESP32_DEV_Onehorse

esp32t.build.f_cpu=240000000L
esp32t.build.flash_mode=dout
esp32t.build.flash_size=4MB

esp32t.menu.FlashFreq.80=80MHz
esp32t.menu.FlashFreq.80.build.flash_freq=80m
esp32t.menu.FlashFreq.40=40MHz
esp32t.menu.FlashFreq.40.build.flash_freq=40m

esp32t.menu.UploadSpeed.921600=921600
esp32t.menu.UploadSpeed.921600.upload.speed=921600
esp32t.menu.UploadSpeed.115200=115200
esp32t.menu.UploadSpeed.115200.upload.speed=115200
esp32t.menu.UploadSpeed.256000.windows=256000
esp32t.menu.UploadSpeed.256000.upload.speed=256000
esp32t.menu.UploadSpeed.230400.windows.upload.speed=256000
esp32t.menu.UploadSpeed.230400=230400
esp32t.menu.UploadSpeed.230400.upload.speed=230400
esp32t.menu.UploadSpeed.460800.linux=460800
esp32t.menu.UploadSpeed.460800.macosx=460800
esp32t.menu.UploadSpeed.460800.upload.speed=460800
esp32t.menu.UploadSpeed.512000.windows=512000
esp32t.menu.UploadSpeed.512000.upload.speed=512000

esp32t.menu.DebugLevel.none=None
esp32t.menu.DebugLevel.none.build.code_debug=0
esp32t.menu.DebugLevel.error=Error
esp32t.menu.DebugLevel.error.build.code_debug=1
esp32t.menu.DebugLevel.warn=Warn
esp32t.menu.DebugLevel.warn.build.code_debug=2
esp32t.menu.DebugLevel.info=Info
esp32t.menu.DebugLevel.info.build.code_debug=3
esp32t.menu.DebugLevel.debug=Debug
esp32t.menu.DebugLevel.debug.build.code_debug=4
esp32t.menu.DebugLevel.verbose=Verbose
esp32t.menu.DebugLevel.verbose.build.code_debug=5
```
Thanks
Mike

me-no-dev, 2017-04-28:
@arcao did you confirm the info above with a board of your own or did you just create a PR?

mjs513, 2017-04-28:
@me-no-dev Some additional info from the description.  It looks like a esp32 dev board except for the DOUT.  I have used this for the module with no problem.

> The SPI embedded flash for some reason requires using the DOUT flash mode instead of the default DIO mode. If this is not an option in your IDE version you have to go into the boards.txt file and add it. 
> 
> This is a small (0.7 in. x 1.4 in.) development board for the ESP32 which includes a 4 MByte N25Q032A SPI NOR flash and an FTDI FT230XQ USB-to-serial converter so Arduino programs can be flashed via the USB Micro-B connector. The ESP32 has Xtensa® Dual-Core 32-bit LX6 microprocessors running at up to 240 MHz achieving up to 600 DMIPS with 448 KByte ROM, 520 KByte SRAM, and 16 KByte SRAM in RTC. The board has a MAX1555 LiPo battery charger, a 3.3 V MIC5528 LDO supplying 500 mA for plenty of power, has reset and boot buttons for easy programming, has a blue led on GPIO5, red led for charging indication, and yellow and orange leds for RX/TX.

me-no-dev, 2017-04-28:
nice layout, but why dout? the chip is QIO based on the name, so why not QIO?

mjs513, 2017-04-28:
Works real nice.  Don't know why dout.  Kris Winer (aka Onehorse) is the owner of the board and recommended dout so I just followed instructions.  You want I should try qio to see what happens?

me-no-dev, 2017-04-28:
yes please :) if it works QIO, then it should stay as the dev board. I kinda not like having gazillion of boards with the exact same config :)

BTW How can I get a board like this?

mjs513, 2017-04-28:
I will give it a try a little later today if you don't mind.  I don't blame you for not wanting a million boards if not necessary.  

You can get one on @kriswiner tindie site, https://www.tindie.com/products/onehorse/esp32-development-board/

igrr, 2017-04-28:
Based on the troubleshooting thread about this board on esp32.com, I remember that DOUT is the only mode which works.

mjs513, 2017-04-28:
In prep for the qio test I decided to just upload blink using DOUT (Arduino ide 1.8.2 and latest esp32 lib) and got the following error:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:2036
load:0x40078000,len:9988
load:0x40080000,len:252
entry 0x40080034
E (30) esp_image: image at 0x1000 has invalid magic byte
E (30) boot: failed to load bootloader header!
user code done
```
I then tried it with 1.8.1 and a previous version of the lib and it worked.  Now what?

Thanks
Mike

me-no-dev, 2017-04-28:
what speed did you use to upload? re-upload might help

me-no-dev, 2017-04-28:
tindie site tells me to join some waiting list :) I guess very few have this board.

mjs513, 2017-04-28:
Its at 92100, just changed it to 115200 and same thing happened.  Tried reloading it a few times and same thing.

Actually, think he sold out of stock.  He had them a couple of weeks ago.  A few of us having been recommending it.  I did it in association with my FreeIMU library so I think it may be getting popular.

mjs513, 2017-04-28:
I used 1.8.1 and the earlier version of the esp32 library and the only mode that worked was dout.  QIO and DIO did not work.  Also tried a couple other speeds and did not work either, got same error message.  I am using esptool - beta version.  Would that make a difference?

kriswiner, 2017-04-28:
Sorry to come late on this thread.

I have sold 14 of these hand-built ESP32 development boards since November,
I think. The first production build of 100 is finishing up and will be
delivered after the Labor Day Holiday in China. I waited until the "new
silicon" was available. In any case, I doubt I will be selling hundreds so
this will always be a small community of users considering the board's
price ($24.95) which is way more than the ESP32 development boards one can
get on EBay. But of course my design is much more useful and worth it!

The SPI flash used on the board seems to require DOUT for programming,
which is incompatible with the default DIO used in the Arduino core. Igrr
created a special bootload.bin file for me which I believe uses DOUT
instead (among other things?, maybe Igrr will remember) and this allows the
board to work normally. I usually use the generic ESP32 board in the board
manager which works fine.

I am not sure a separate board entry is required; there is no downside to
using the generic board selection. I do wish the boot load problem would be
fixed in the Arduino core though so a separate version of bootload.bin
wouldn't have to be used. If it would help, I would be happy to send a
board for evaluation. Just let me know where nd to whom.

me-no-dev, 2017-04-28:
@kriswiner separate bootloader can be added for your board if that is required, but since you say that you use the "generic ESP32 board" then I wonder why you need separate bootloader? If you need to upload in dout only and the flash can run in qio once booted, maybe we can do something about that too. BTW I love the form factor! The only thing I would have changed is incorporate UFL pads by the antenna pin so UFL could be soldered and external antenna used if wanted.

I will surely give you my address :) I don't mind paying for it either ;)
Please email me at hristo@espressif.com, so I do not need to publish my address here :)

mjs513, 2017-04-28:
@kriswiner - just posted a pr on your library.  As far I can see, I don't have an option for an esp32 generic in my dropdown, just the esp32 dev board.

mjs513, 2017-04-28:
@me-no-dev To test whether the problem is with the ESP32 library or the IDE I copied the previous version of the library that I was using into the 1.8.2 hardware folder and uploaded the Blink sketch.  It worked fine no problems.  Yes it blinks.  Not sure what version I have, but the json file is dated 2/4/17 along with most of the files.  The make file is dated 2/20 if that helps.  

You want a separate PR?

mjs513, 2017-04-30:
Closing this since the board file has been uploaded and is identified

closed 2017-04-30 by mjs513