espressif/arduino-esp32 #335: bluetooth a2dp support

h3ndrik, 2017-04-26:
Hi all,

I discovered the bluetooth a2dp example in the esp-idf: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/a2dp_sink
however arduino is complaining about some missing header files.
Is it possible to add a2dp support, or am i not supposed to mix arduino code with espressif framework stuff?

ShendoXT, 2017-04-27:
I have played with ESP32 for a few days and this my current understanding of the things:
Arduino is just a layer over esp-idf so example code should work in arduino.

When compiling the sample code you first have to configure build options using menuconfig
and it seems like bluetooth support is turned off by default.
You should check what options are configured for arduino builds.

What is the exact error message you got?

closed 2017-05-19 by me-no-dev

Elektrik1, 2017-08-12:
Any chance a2dp example will be ported in arduino examples ? I have tried to port it manually, ended up with ISR errors in serial output, seems like callbacks are not transfered correctly.

me-no-dev, 2017-08-17:
here is the example as running on Arduino, though I get exceptions and other issues when trying to connect from a computer/phone. Those might be related to stack being too small on the current build. But I can not rebuild Arduino, because there are some other BT bugs introduced in recent IDF
[A2DP_Test.zip](https://github.com/espressif/arduino-esp32/files/1231768/A2DP_Test.zip)

igrr, 2017-08-17:
But you can rebuild the old IDF with the larger stack version in menuconfig :)

me-no-dev, 2017-08-17:
Not much point into rebuilding ad releasing if the only gain would be this stack size :) and given that the current BT issues would be fixed soon (will they?)

matasondrak, 2017-08-20:
Looking forward to the Arduino BT implementation and can't wait! BT is kind of unusable for now

Elektrik1, 2017-09-15:
All working fine now with latest IDF build. Thanks!

Elektrik1, 2017-10-14:
After upgrading to latest, same code started to not to work again. I assume IDF was built without BT again. Errors:
```
Sketch.ino.cpp.o:(.literal.bt_av_hdl_stack_evt+0x10): undefined reference to `esp_a2d_register_callback'

Sketch.ino.cpp.o:(.literal.bt_av_hdl_stack_evt+0x14): undefined reference to `esp_a2d_register_data_callback'

Sketch.ino.cpp.o:(.literal.bt_av_hdl_stack_evt+0x18): undefined reference to `esp_a2d_sink_init'

Sketch.ino.cpp.o:(.literal.bt_av_hdl_stack_evt+0x1c): undefined reference to `esp_avrc_ct_init'

Sketch.ino.cpp.o:(.literal.bt_av_hdl_stack_evt+0x20): undefined reference to `esp_avrc_ct_register_callback'

Sketch.ino.cpp.o:(.literal.bt_av_hdl_stack_evt+0x24): undefined reference to `esp_bt_gap_set_scan_mode'

Sketch.ino.cpp.o:(.literal._Z4loopv+0xc): undefined reference to `esp_avrc_ct_send_passthrough_cmd'
```

williamesp2015, 2018-07-19:
Seems it is done.
https://github.com/dgiebert/techfest17/blob/master/A2DP_Test/A2DP_Test.ino

colmakers-antonio, 2018-07-26:
Hi, i have a problem using this version for arduino, i got this error, someone can help me?

`'esp_a2d_register_data_callback' was not declared in this scope`

tueddy, 2018-07-30:
Change to ..esp_a2d_sink_register_data_callback.. ?

ronyeapen, 2019-02-11:
Hello guys, I am getting errors as noted by @Elektrik1. I am not sure how @williamesp2015 's solution is any different to the code @me-no-dev put up. What exactly is the solution to this problem? thanks for your help.

Petros144, 2019-03-31:
also have the problem that it wont work, ESP just prints some wired gibberish and than reboots / hangs...

chenbhil, 2020-03-16:
Still getting linker error here, I'll be thankfull for any kind of help.
`.pio\build\esp32dev\src\main.o:(.literal.bt_av_hdl_stack_evt+0x1c): undefined reference to `esp_a2d_register_callback'
.pio\build\esp32dev\src\main.o:(.literal.bt_av_hdl_stack_evt+0x20): undefined reference to `esp_a2d_sink_register_data_callback'
.pio\build\esp32dev\src\main.o:(.literal.bt_av_hdl_stack_evt+0x24): undefined reference to `esp_a2d_sink_init'
.pio\build\esp32dev\src\main.o:(.literal.bt_av_hdl_stack_evt+0x28): undefined reference to `esp_avrc_ct_init'
.pio\build\esp32dev\src\main.o:(.literal.bt_av_hdl_stack_evt+0x2c): undefined reference to `esp_avrc_ct_register_callback'
.pio\build\esp32dev\src\main.o:(.literal.bt_av_hdl_stack_evt+0x30): undefined reference to `esp_bt_gap_set_scan_mode'
.pio\build\esp32dev\src\main.o: In function `bt_av_hdl_stack_evt':
main.c:(.text.bt_av_hdl_stack_evt+0x12): undefined reference to `esp_a2d_register_callback'
main.c:(.text.bt_av_hdl_stack_evt+0x1b): undefined reference to `esp_a2d_sink_register_data_callback'
main.c:(.text.bt_av_hdl_stack_evt+0x21): undefined reference to `esp_a2d_sink_init'
main.c:(.text.bt_av_hdl_stack_evt+0x27): undefined reference to `esp_avrc_ct_init'
main.c:(.text.bt_av_hdl_stack_evt+0x30): undefined reference to `esp_avrc_ct_register_callback'
main.c:(.text.bt_av_hdl_stack_evt+0x39): undefined reference to `esp_bt_gap_set_scan_mode'`

lbernstone, 2020-03-17:
Your header doesn't match your library.  Don't hijack issues, particularly closed ones.