espressif/arduino-esp32 #340: Tools -> Board: the section ESP32 was disappeared.

kendo55, 2017-04-30:
I installed the latest Arduino IDE v 1.8.2 with the Windows Installer.
And then ESP32 Support, like described under:
https://github.com/espressif/arduino-esp32/blob/master/doc/windows.md
.
it works fine!!!

Then i had to test some STM32 board.
For that, i selected the "boards manager".
Where i selected the "STM32F4 Boards by Aduino"
and installed it.

After that step, the section ESP32 was disappeared, 
in the menu: Tools -> Board

It raises the question, where is the definition file,
where we can tell the Arduino IDE that we are using ESP32.

The same question is also arises in issue #328

me-no-dev, 2017-04-30:
Did arduino erase you esp32 folder? Where did you put it? I see no reason why board will disappear once you install other boards. They go into totally separate folders all together.

mjs513, 2017-04-30:
After I installed 1.8.2 I had to re-install the esp32 into the 1.8.2 hardware directory.  You can also just copy the esp32 into the hardware directory of 1.8.2, that works as well.

closed 2017-05-19 by me-no-dev