espressif/arduino-esp32 #341: Watchdog with System Reset

PRosenb, 2017-04-30:
According to issue #147, the watchdog functionality of the ESP32 is available to Arduino sketches.

In order to make sure, the sketch is always functional, I want to use the watchdog to reset the chip if it gets stuck somehow. When I call `esp_task_wdt_feed()` once and then block the `loop()` thread, nothing happens though.

According to the [Task watchdog](http://esp-idf.readthedocs.io/en/latest/api-reference/system/wdts.html) documentation, the watchdog is supposed to issue a warning but not reset the CPU. For Arduino sketches, I may be better to reset the system instead. Is that possible without manipulating the related watchdog registers directly?

me-no-dev, 2017-05-06:
The reason is that WDT is disabled for the idle task on CPU1 (Arduino setup/loop). If you want to have all options, you might want to look into using Arduino as IDF component.

closed 2017-05-19 by me-no-dev

PRosenb, 2017-05-19:
Thanks @me-no-dev, I'll do that.

dmody, 2017-06-12:
I'm sorry guys, I'm a not familiar with programming the ESP32 without using the Arduino IDE, and I don't understand the solution proposed.   

If I understand what you're saying here it is that:
* watchdogs can not be used for the setup and loop routines in a Arduino.  This is because they have been disabled for those two particular tasks.

*I think you're suggesting, but I'm definitely not sure, that you need to switch to programming using Eclipse (the preferred programming environment instead of the Arduino IDE), and use the Arduino code as a component.  

Hopefully you can set me straight about watchdogs.    What I'm finding is that my ESP32 is 'randomnly' halting, or it's gone into an infinite loop, or it's not restarting from deep sleep, and  I would like to use a watchdog to at least eliminate one of those alternatives.

PRosenb, 2017-06-12:
Hi @dmody , @me-no-dev refers to building the code as described here:
https://github.com/espressif/arduino-esp32#using-as-esp-idf-component
You don't need Eclipse for it, any text editor will do.

dmody, 2017-06-13:
Thank you. I'll read up on that.

inforaudio, 2017-08-17:
when install idf, we can compile with diferent options in espressif. But, how can use if in Arduino.?

me-no-dev, 2017-08-17:
@inforaudio you can not set those when using Arduino stand-alone ;) you can if you use it as IDF component (answer is two positions above your question)

mxwang101, 2017-08-17:
To All, Same problem, please give us step by step how to make Watchdogs work as use the Arduino code as a component?

mxwang101, 2017-08-18:
Setup " void esp_int_wdt_init();"  OK.  But no ideal how to use it in ESP32?  a demo prgram code will big help.

#include <CurieBLE.h>
#include <esp_task_wdt.h>
#include <esp_int_wdt.h>

int ctime=1;

void esp_int_wdt_init();

void setup() {
  Serial.begin(9600);
  while (!Serial);

  // initialize the BLE hardware
  BLE.begin();

----------Above OK , works

hcs-svn, 2017-08-18:
> The reason is that WDT is disabled for the idle task on CPU1 (Arduino setup/loop)

Will we get this for arduino-esp32 or what is the reason why we don't have/get it?

mxwang101, 2017-08-18:
A simple demo code will fix it, ESP team please build one, Thanks

mxwang101, 2017-08-18:
A WDT demo code will fix all problems. ESP team, please release one. Just like 51 ,  TI430, ,,, all have WDT demo codes. Thanks

inforaudio, 2017-08-18:
We have a bigger project finished with ESP32 and now we find it¡¡¡.. 
A very bigger problem because without a Watchdog, we can use it in production.

@mxwang101  we have problem with your code.  Really run it properly?
@me-no-dev  where can i see how use IDF for configure ESP32 Arduino ide with watchdog. (whit details)?
all can i find is for compile espressif proyect.

are there a compilation / libraries /  nre version where watchdog is enabled??  So complicated is it?  We pass 2 days looking for and we are thinking in use other ESP version

me-no-dev, 2017-08-18:
@inforaudio all you need to do is to install IDF, install Arduino as IDF component and run ```make menuconfig``` in your project folder in order to access the build config menu.
Info: https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

inforaudio, 2017-08-18:
how can install Arduino as IDF component?
"make menuconfig in your project folder", in my arduino folder sketch?? .ino?  or where? ( here dont run )

after it?? what can we do.?
in the future can we continue compile the proyect with Arduino IDE  or with make flash? 
do you know some tutorial about it?  text, video etc..

lrmoreno007, 2017-08-18:
https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

inforaudio, 2017-08-19:
Thanks for your help. 
So i think that we have some advance.

We create a idf project (copy hello world example) , 
we made components folder,  clone git.. and make menuconfig,  and activate the arduino options.
Then in main.cpp copy my code (Arduino code).
This time is:  (the example).
//file: main.cpp
#include "Arduino.h"

void setup(){
  Serial.begin(115200);
}

void loop(){
  Serial.println("loop");
  delay(1000);
}

try to made a make and have this problem
make[1]: *** No hay ninguna regla para construir el objetivo 'cores/esp32/main', necesario para 'libarduino.a'.  Alto.
make: *** [/home/Pablo/esp/esp-idf/make/project.mk:391: component-arduino-build] Error 2

What can i do now? What is the problem?

usmanshahid001, 2017-08-31:
hiii
i am working in automation company and we need fail safe operation we need a wdt which could reset the whole system in case of any crash but i am unable to find any RWDT documentation yet kindly develop library in this regard
thanks

assegid, 2017-11-16:
This issue should not have been closed. It was not properly addressed. None of the explanations given above are complete or seem to work. I think a simple wdt solution in the Arduino IDE environment should be made available. The current solution seems to be if you need the wdt use the IDF. What good is an IOT device that gets deployed and goes dead in 2 days because of some obscure bug. In my opinion if the development team cannot provide a simple wdt solution for the Arduino IDE a very important feature is being ignored.

usmanshahid001, 2017-11-16:
Yeah i am so disappointed as well.Development is useless without wdt

CarlosGS, 2017-12-13:
I don't fully understand the implications of this. Does it mean that any Arduino sketch for ESP32 will never be able to automatically reset when stuck?

unixbigot, 2018-11-27:
For those like me who tripped over this thread as a high-ranking hit on search for "arduino esp watchdog", it does now work.   Example here: https://github.com/unixbigot/esp-skeleton

PriyankPa, 2018-12-28:
I can confirm that the current Arduino ESP32 SDK does not reset the device after watchdog is triggered.

me-no-dev, 2018-12-28:
how about I add API to enable WDT on the idle task and on the loop task? will that satisfy?

me-no-dev, 2018-12-28:
API added and WDT reboot enabled in https://github.com/espressif/arduino-esp32/pull/2248

usmanshahid001, 2018-12-28:
that's really good