espressif/arduino-esp32 #342: Rcving Error: "E (30) esp_image: image at 0x1000 has invalid magic byte E (31) boot: failed to load bootloader header!"

mjs513, 2017-04-30:
Downloaded latest board files into IDE 1.8.2 and receiving the following error messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:2036
load:0x40078000,len:9988
load:0x40080000,len:252
entry 0x40080034
E (30) esp_image: image at 0x1000 has invalid magic byte
E (31) boot: failed to load bootloader header!
user code done

```
This first mentioned in https://github.com/espressif/arduino-esp32/issues/334 .  Tried different upload speeds and did not work.  

To test whether the problem is with the ESP32 or the IDE I copied the previous version of the library that I was using into the 1.8.2 hardware folder and uploaded the Blink sketch. It worked fine no problems. Yes it blinks. Not sure what version I have, but the json file is dated 2/4/17 along with most of the files. The make file is dated 2/20 if that helps.

Thank you
Mike

me-no-dev, 2017-05-06:
so you are not sure if you are running the latest code? Did you resolve this?

mjs513, 2017-05-06:
The problem surfaced with the latest code.  The old code dated 2/20 worked but the last version I downloaded didn't.  I see you just did an update for the chip id.  I will download as of today and give it another run.

me-no-dev, 2017-05-06:
i'll have in a few minutes a new update with new bootloader and all. Please give it a go once it's merged (5-10 minutes tops)

mjs513, 2017-05-06:
Not a problem.  Will try later today and let you know how it works.

me-no-dev, 2017-05-06:
It's merged :) https://github.com/espressif/arduino-esp32/commit/376961d16844a3190bfc409e74c6acd0a4c93224

mjs513, 2017-05-06:
No luck.  Still getting the same error.  You want I zip the version that works?

me-no-dev, 2017-05-06:
ok what board are you using exactly?

mjs513, 2017-05-06:
Onehorse's ESP32 board.

me-no-dev, 2017-05-06:
do you have that board selected in the menu? that one has specific issues with flashing ;)

mjs513, 2017-05-06:
Yes I have that specific board selected in the menu. Had no problem when using the older version of the ESP32 Lib.  Something seem to have changed between versions I am using.  I first mentioned this in issue #334.

Here is output from the IDE:

```
Archiving built core (caching) in: C:\Users\CYBERP~1\AppData\Local\Temp\arduino_cache_811494\core\core_espressif_ESP32_onehorse32dev_FlashFreq_80,UploadSpeed_921600_23f4c7672b1613cabd9907ba53176f8b.a
Sketch uses 104942 bytes (10%) of program storage space. Maximum is 1044464 bytes.
Global variables use 9500 bytes (3%) of dynamic memory, leaving 285412 bytes for local variables. Maximum is 294912 bytes.
esptool.py v2.0-beta3
Connecting.......
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash params set to 0x032f
Compressed 14304 bytes to 8831...

Writing at 0x00001000... (100 %)
Wrote 14304 bytes (8831 compressed) at 0x00001000 in 0.1 seconds (effective 1049.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 105...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (105 compressed) at 0x00008000 in 0.0 seconds (effective 1536.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 3449.3 kbit/s)...
Hash of data verified.
Compressed 188240 bytes to 58130...

Writing at 0x00010000... (25 %)
Writing at 0x00014000... (50 %)
Writing at 0x00018000... (75 %)
Writing at 0x0001c000... (100 %)
Wrote 188240 bytes (58130 compressed) at 0x00010000 in 4.4 seconds (effective 342.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
```

kriswiner, 2017-05-06:
The first production batch of ESP32 boards was shipped yesterday so I will be able to send me-no-dev one next week or so. Maybe this would be easier than remote debugging. I haven't updated the ESP32 Arduino core (afraid to!) since the first time I got the board to work using Igrr's bootload.bin file. I am sure once me-no-dev has the board the problem will get sorted out in short order.

mjs513, 2017-05-06:
Thanks @kriswiner .  Just wanted to let you know the issue with the new core.

@me-no-dev and @kriswiner. As a hack I took the bootloader.bin file from the version that works and passed it in the one I just downloaded and it worked no problem.  Didn't think about doing this until @kriswiner mentioned the bootload.bin file in your post. At least I have a workaround for the new core.  Not sure what changed.

me-no-dev, 2017-05-19:
I'm waiting for the board :) will do my best once here!

mjs513, 2017-05-19:
Thanks @me-no-dev. At least I found a work around to be able to use the new core, i.e., just use the old bootloader :)

kriswiner, 2017-05-19:
Yeah, that's what I have been doing, but I would rather be able to use the
stock Arduino core as it is. (and get updates through the Arduino IDE json!)

RSchlenker, 2017-05-29:
@mjs513 could you somehow publish the actual version of the bootloader you are using?

I also have issues with the bootloader.bin from @kriswiner 

I got a different log though:

> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> ets Jun  8 2016 00:22:57
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0x00
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DOUT, clock div:1
> load:0x3ffc0008,len:0
> load:0x3ffc0008,len:3868
> load:0x40078000,len:5028
> load:0x40080000,len:260
> entry 0x40080034
> e[0;32mI (17) boot: Espressif ESP32 2nd stage bootloader v. V0.1e[0m
> e[0;32mI (23) boot: compile time 18:15:32e[0m
> D (27) esp_image: reading image header @ 0x1000e[0m
> D (31) boot: magic e9e[0m
> D (33) boot: segments 04e[0m
> D (36) boot: spi_mode 03e[0m
> D (38) boot: spi_speed 0fe[0m
> D (41) boot: spi_size 02e[0m
> e[0;32mI (43) boot: SPI Speed      : 80MHze[0m
> e[0;32mI (47) boot: SPI Mode       : DOUTe[0m
> e[0;32mI (51) boot: SPI Flash Size : 4MBe[0m
> e[0;32mI (55) boot: Partition Table:e[0m
> e[0;32mI (59) boot: ## Label            Usage          Type ST Offset   Lengthe[0m
> D (66) bootloader_flash: mmu set paddr=00000000 count=1e[0m
> D (71) boot: mapped partition table 0x8000 at 0x3f408000e[0m
> D (76) boot: load partition table entry 0x3f408000e[0m
> D (81) boot: type=1 subtype=2e[0m
> e[0;32mI (83) boot:  0 nvs              WiFi data        01 02 00009000 00005000e[0m
> D (91) boot: load partition table entry 0x3f408020e[0m
> D (95) boot: type=1 subtype=0e[0m
> e[0;32mI (98) boot:  1 otadata          OTA data         01 00 0000e000 00002000e[0m
> D (105) boot: load partition table entry 0x3f408040e[0m
> D (110) boot: type=0 subtype=10e[0m
> e[0;32mI (113) boot:  2 app0             OTA app          00 10 00010000 00100000e[0m
> D (120) boot: load partition table entry 0x3f408060e[0m
> D (125) boot: type=0 subtype=11e[0m
> e[0;32mI (128) boot:  3 app1             OTA app          00 11 00110000 00100000e[0m
> D (136) boot: load partition table entry 0x3f408080e[0m
> D (140) boot: type=1 subtype=82e[0m
> e[0;32mI (143) boot:  4 spiffs           Unknown data     01 82 00210000 001f0000e[0m
> D (151) boot: load partition table entry 0x3f4080a0e[0m
> D (155) boot: type=ff subtype=ffe[0m
> e[0;32mI (159) boot: End of partition tablee[0m
> D (163) bootloader_flash: mmu set paddr=00000000 count=1e[0m
> e[0;32mI (168) boot: Loading app partition at offset 00010000e[0m
> D (173) esp_image: reading image header @ 0x10000e[0m
> D (178) esp_image: reading 8 image segmentse[0m
> D (182) bootloader_flash: mmu set paddr=00010000 count=2e[0m
> D (244) bootloader_flash: mmu set paddr=00020000 count=1e[0m
> D (256) bootloader_flash: mmu set paddr=00020000 count=1e[0m
> D (264) bootloader_flash: mmu set paddr=00020000 count=1e[0m
> D (265) bootloader_flash: mmu set paddr=00020000 count=1e[0m
> D (291) bootloader_flash: mmu set paddr=00020000 count=1e[0m
> D (292) bootloader_flash: mmu set paddr=00020000 count=2e[0m
> D (306) bootloader_flash: mmu set paddr=00030000 count=1e[0m
> D (357) esp_image: reading checksum block at 0x3dff0e[0m
> D (357) esp_image: reading image header @ 0x10000e[0m
> D (357) boot: bin_header: 233 8 3 2 4008103ce[0m
> e[0;32mI (360) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) e[0m
> D (368) boot: found drom segment, map from 00020010 to 3f400010e[0m
> e[0;32mI (373) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x032e8 ( 13032) mape[0m
> e[0;32mI (381) boot: segment 2: paddr=0x000232f8 vaddr=0x3ffc0000 size=0x01c54 (  7252) loade[0m
> D (389) bootloader_flash: mmu set paddr=00020000 count=1e[0m
> Guru Meditation Error of type IntegerDivideByZero occurred on core  0. Exception was unhandled.
> Register dump:
> PC      : 0x400ddb9d  PS      : 0x00060f30  A0      : 0x80086a49  A1      : 0x3ffe3bb0  
> A2      : 0x779fffff  A3      : 0x00000009  A4      : 0x00000000  A5      : 0x00000000  
> A6      : 0x00000009  A7      : 0x00000000  A8      : 0x84008000  A9      : 0x00000009  
> A10     : 0x00000009  A11     : 0x00000001  A12     : 0x779fffff  A13     : 0x00000000  
> A14     : 0x00000000  A15     : 0x3ffc0288  SAR     : 0x00000012  EXCCAUSE: 0x00000006  
> EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
> 
> Backtrace: 0x400ddb9d:0x3ffe3bb0 0x40086a49:0x3ffe3bd0 0x400d56e0:0x3ffe3bf0 0x400d56fd:0x3ffe3c10 0x40080f25:0x3ffe3c30 0x40081158:0x3ffe3c60 0x4007859d:0x3ffe3ca0 0x400787e9:0x3ffe3cd0 0x40078c45:0x3ffe3d50 0x40080101:0x3ffe3e70 0x40007c34:0x3ffe3eb0 0x40000740:0x3ffe3f20
> 
> CPU halted.

I am not sure what "Guru Meditation Error of type IntegerDivideByZero" means.
I also used the Onehorse Board and downloaded the most recent arduino core 2 days ago.

Kind Regards,
Robin

mjs513, 2017-05-29:
@RSchlenker Here's the bootloader that I am using.  Good luck.  Let us know if it works.  Might help @me-no-dev.

Mike
[bootloader.zip](https://github.com/espressif/arduino-esp32/files/1035851/bootloader.zip)

me-no-dev, 2017-05-29:
Sill waiting the post to bring the board :) fingers crossed it will be soon

RSchlenker, 2017-06-02:
@mjs513 Thank you for the bootloader.
It saved my day so far! For me, your version of the bootloader worked and I was able to flash the board successfully.
@kriswiner You could think about replacing the bootloader in your repository with the one provided by mike

Robin

chevdor, 2017-07-10:
I am having the same issue.
I am using a very simple blink code that works fine on a ESP32-DevKitC.
I get the error however when trying to flash an ESP32-WROVER-KIT.

mjs513, 2017-07-17:
@chevdor try the bootloader version in a previous post.  Maybe it will help?

tobozo, 2017-07-17:
@chevdor select 40Mhz in the board menu instead of 80Mhz and your blink code should work

chevdor, 2017-07-18:
@mjs513 @tobozo Thank you guys, indeed 40Mhz DIO does the trick.

closed 2018-08-05 by copercini

D-Ace-22, 2024-05-24:
if the selected flash mode is QIO in the previous firmware then you will not be able to do the OTA on it so use the flash mode : DIO and i am working with 80Mhz and so there will be no effect of frequency on OTA