espressif/arduino-esp32 #344: Type comparison is always false due to limited range of data type [-Wtype-limits]

ESP32DE, 2017-04-30:
hi

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/libb64/cdecode.c#L15
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/libb64/cdecode.c#L45
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/libb64/cdecode.c#L55
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/libb64/cdecode.c#L66
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/libb64/cdecode.c#L77

best wishes
rudi ;-)

me-no-dev, 2017-05-06:
Yes I've seen those and I wonder what they are about. The library is working fine though. Char is signed AFAIK, so I really do not get the reason for the warnings.

me-no-dev, 2017-05-06:
Nevermind :) found it

closed 2017-05-06 by me-no-dev