espressif/arduino-esp32 #348: Little problem in BasicOTA.ino without md5 password...

lrmoreno007, 2017-05-03:
Hi, there is a little problem using the BasicOTA.ino example. There are 3 ways to identify yourself:
- Without password
- With password
- With MD5 password

When using the first and second option, you get the error: [ERROR]: 1460. But interestingly the sketch has been loaded and has overwritten the previous one perfectly, I've checked it by including a variable that increases the version number. Therefore the problem is only that it gives the process as error and the red sign appears in Arduino.

When I use the third option (MD5) everything is perfect.

Regards and thanks for everything.
----------------------------------------------------------------------------------------------------
- Git version
- ESP32-ST (WEMOS Lolin32 is selected)
[image]

me-no-dev, 2017-05-03:
Do not mind this error ;) I know about it and just have not properly caught the reason for it. Sketch is uploaded OK unless you get a more meaningful error. 1460 is the size of the packets it got, so all is good ;)

closed 2017-05-03 by me-no-dev

lrmoreno007, 2017-05-19:
It's solved with the last git version.

Thanks.