espressif/arduino-esp32 #351: SimpleWiFi server example connection reset/crash

mindThomas, 2017-05-04:
Trying out the SimpleWiFi server example on a brand new eBox&Widora EPS32 module today.
Unfortunately it does not seem to run out of the box. When accessing the webpage it seem to be served the first time but then the ESP crashes. See the serial output below:
```
Connecting to default
...
WiFi connected
IP address: 
192.168.1.20
new client
client disonnected
new client
client disonnected
new client
client disonnected
E (575908) wifi: pp.c 3221

new client
client disonnected
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: ppT
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not feed the watchdog in time:
Tasks currently running:
CPU 0: ppT
CPU 1: IDLE
```

Do you have success with the Webserver example?
Do I need to configure the ESP in any way not done through upload (eg. makeconfig?)

What does this line mean in the output: `E (575908) wifi: pp.c 3221`
It seems like some sort of error code, but is pp.c referring to a file?

me-no-dev, 2017-05-05:
Are you running through IDF?

mindThomas, 2017-05-05:
No. I'm using the Arduino IDE in where I have installed the arduino-esp32 tools package and then tried out the "SimpleWiFi server" example

me-no-dev, 2017-05-05:
Those watchdog messages are the reason I ask. WDT is disabled for the IDLE task on CPU1 and I see it mentioned up there.
BTW I have no issues with the example on my end.

mindThomas, 2017-05-05:
I have just opened the tcp_perf being part of the ESP-IDF and I seem to get the same errors with this project when I send a lot of packages to the chip.
Running a "make menuconfig" on this and going into the component specific settings I see that the IDLE WDT is enabled.

me-no-dev, 2017-05-19:
Did you resolve this? BTW IDLE WDT is generally enabled, just not on the CPU that runs Arduino.

mindThomas, 2017-05-19:
No, I did not get this to work. I kept running into too many problems with the ESP32 and the current state of this Arduino project.
I decided to switch to the ESP8266 instead which seemed to be able to provide me with reasonable performance after several tuning attempts. Now I am able to transfer files to the ESP8266 with ~10 Mbit.

copercini, 2017-08-25:
@mindThomas This issue was solved? Can you post some code?

lonerzzz, 2017-08-30:
@me-no-dev Thinking we should close this one unless we have a path to otherwise reproduce?

me-no-dev, 2017-08-30:
@lonerzzz I agree :)

closed 2017-08-30 by me-no-dev