espressif/arduino-esp32 #352: AnalogOut example

zhivko, 2017-05-04:
When trying DAC example I get strange noisy up to 5V signal on pin 18.
Sample I tried:
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/AnalogOut/SigmaDelta/SigmaDelta.ino

You can see scope measurements on pin 18 in this google album:
https://goo.gl/photos/7pRuDPMRH8GQYbB9A

I also tried changing pin 18 to pin 25 in 
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/AnalogOut/SigmaDelta/SigmaDelta.ino#L6
(according to http://esp32.de/ESP32_Pinout_a1_3.png dac is on gpio25) but it happens same on that pin.

zhivko, 2017-05-05:
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/AnalogOut/LEDCSoftwareFade/LEDCSoftwareFade.ino
actually worked as you described but I was expecting analog out on pin - is there real dac on ESP32 that would generate 0.3.3V on pin with 0 - 255 discrete voltages in between?

me-no-dev, 2017-05-05:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-dac.h#L29

closed 2017-05-05 by me-no-dev