espressif/arduino-esp32 #353: WiFiSTAClass::setAutoReconnect from WiFiSTA.h is not implemented

bmitov, 2017-05-04:
WiFiSTAClass::setAutoReconnect method from WiFiSTA.h file is declared but not implemented in the .cpp file

me-no-dev, 2017-05-22:
Implemented :) sort of...

closed 2017-05-22 by me-no-dev

St3venAU, 2017-05-24:
I thought I'd give this a test, and so far it doesn't look like it does an automatic reconnect. This is my test code:

https://gist.github.com/St3venAU/8367e1e8b0a267ae558326aaf9888c94

When I leave the manual reconnect code commented out, and I turn off the wifi on my router, the status goes to CONNECTION_LOST and stays that way even after turning the wifi back on.

With the manual reconnect code un-commented, the status changes to CONNECTION_LOST, then NO_SSID_AVAIL, and then finally CONNECTED again once the wifi signal is returned.

ogiewon, 2017-09-06:
@me-no-dev I have noticed the exact same thing.  The ESP32 does not seem to implement any sort of automatic WiFi reconnect logic.  The ESP8266 does this very well, behind the scenes of the Arduino developer.  I would really like to see this implemented.  For now, I have to detect the loss of connection and then attempt the WiFi.reconnect() command like @St3venAU (although I trap it in a WiFi Event handler.)

Has there been any progress on this issue since May?  Hoping I am just missing something obvious! ;)

Thanks!

nenadalm, 2018-11-13:
@me-no-dev I don't get it. How is it implemented? It doesn't seems to work. When I checked the code, it's just about setting the flag: https://github.com/espressif/arduino-esp32/commit/b6e83fac62bbb21f032ec45122a43cb5f963bf24 but nothing seems to be using the flag. Yet the commit claims that the behaviour is implemented. Or maybe I misunderstood something. Other issue related to this is also closed: https://github.com/espressif/arduino-esp32/issues/653.

Bmooij, 2019-02-12:
@nenadalm It is used in [WiFiGeneric.cpp](https://github.com/espressif/arduino-esp32/blob/05d72f963d91c94518e6d08e2805a0a078c8af44/libraries/WiFi/src/WiFiGeneric.cpp#L375)