espressif/arduino-esp32 #356: Add update instructions

ralphrmartin, 2017-05-05:
Please edit the readme so that there are instructions not just for installing this for the first time, but also for using git to keep it updated.

Better still, make it all work through the usual arduino update mechanism.

me-no-dev, 2017-05-05:
This is an open project, why not contribute those things yourself?

me-no-dev, 2017-05-05:
BTW such instructions exists in the Windows readme

ralphrmartin, 2017-05-05:
(1) If I knew how to do it, I would contribute it....

(2) Why would a Mac or Linux user read the Windows readme?

Why not spend your time improving things rather than writing unhelpful responses?

me-no-dev, 2017-05-05:
> Why not spend your time improving things rather than writing unhelpful responses?

Huh? You sound like you know what I'm doing? Seems documentation is more important to you than actual coding and features? Seems that you think I can do 100 things at a time? But at the same time you can's do a simple query to Google?
FYI, my answer was in no way offending, but yours...

ralphrmartin, 2017-05-05:
I am happy to believe you are doing some really good development work. But it is now good doing that if no one knows how to use it. Documentation should go hand in hand with development. Please keep calm, and carry on...

closed 2017-05-19 by me-no-dev