espressif/arduino-esp32 #357: Windows Setup - Virus in esptool-9072736-windows.zip?

VisualMicro, 2017-05-06:
Hi,

The windows install instructions failed to create the tools defined in the platform.txt (shown below):-

- gen_esp32part.exe
- espota.exe
- gen_esp32part.exe

I think Symantec is treating the esptool.exe as Trojan.Gen.8lcloud and virustotal.com shows some trojan results. https://www.virustotal.com/en/file/eac14998df3fed9998943ace5c2f07114b678869f143c3316f4ff26cc851e268/analysis/

```
tools.esptool.cmd="{runtime.platform.path}/tools/esptool"
tools.esptool.cmd.linux=python "{runtime.platform.path}/tools/esptool.py"
tools.esptool.cmd.windows="{runtime.platform.path}/tools/esptool.exe"

tools.esptool.network_cmd=python "{runtime.platform.path}/tools/espota.py"
tools.esptool.network_cmd.windows="{runtime.platform.path}/tools/espota.exe"

tools.gen_esp32part.cmd=python "{runtime.platform.path}/tools/gen_esp32part.py"
tools.gen_esp32part.cmd.windows="{runtime.platform.path}/tools/gen_esp32part.exe"

```
**Observations/Questions**

- Should gen_esp32part and espota use the python versions instead of .exe? gen_esp32part.py seems to run without error but I haven't looked at what it should do or if the output is valid.
- Esptool.exe (v0.4.9) from the esp8266 install does not cause a false positive but doesn't appear to work with the esp32
 

Thanks

me-no-dev, 2017-05-06:
python scripts require python and some extra modules installed. We opted for the compiled versions for windows because of many issues with installations and cases where people have python3. You should tell your Symantec to just ignore them :) Seems like Norton is the only software that complains about them

VisualMicro, 2017-05-06:
Thanks. I agree it's annoying. Most people have python installed because the esp8266 uses it. 

Would it be possible for you to remove the message from esptool.py that prevents it running on windows? 

Then, if we have python installed, we can simply override platform.txt.

me-no-dev, 2017-05-06:
> remove the message from esptool.py that prevents it running on windows

what message? all you need is to change platform.txt

VisualMicro, 2017-05-06:
Sorry yes, I had a permissions issue and thought it was .py file. 

It's working nicely thanks for the great work.

Windows users with python installed can create a "C:\Users\<YoruName>\Documents\Arduino\Hardware\espressif\esp32\platform.local.txt" with the following lines:-

```
tools.esptool.cmd.windows=python "{runtime.platform.path}/tools/esptool.py"
tools.esptool.network_cmd.windows=python "{runtime.platform.path}/tools/espota.exe"
tools.gen_esp32part.cmd.windows=python "{runtime.platform.path}/tools/gen_esp32part.py"
```

closed 2017-05-06 by VisualMicro