espressif/arduino-esp32 #360: ESP32 Examples

anteph, 2017-05-07:
Hi,

I'm creating some tutorials / examples for the ESP32 Arduino support at my blog:
https://techtutorialsx.com/category/esp32/

I'm covering both basic topics and also some more advanced ones, such as the use of some FreeRTOS functionalities. I'm also including some explanations and images of the expected results.

Hope it helps others getting started with ESP32 development.

Best regards,
Nuno Santos

tushaar, 2017-05-07:
I saw your blog. Looks interesting will be useful to me and others. Request you to do tutorials / examples on bluetooth BLE client and server. 

But according to me this should not be created as an issue. You could put it on ESP32 Forum under sample code etc. - https://www.esp32.com

me-no-dev, 2017-05-07:
Maybe we can link the tutorials to the documentation :)

anteph, 2017-05-08:
@tushaar , thank you for the feedback :) As soon as I have some time I will test BLE and try to post some content about it. Right now I'm playing with RTOS and the execution of code in the two cores of the ESP32.

I didn't know where was the best place to post the link, so I opened the issue in hope someone would see it and move it to the appropriate place. Thanks for the recommendation, I will share at the forum. Should I close this issue or the appropriate action is to wait for a collaborator to close it?

@me-no-dev let me know which links you recommend me to include in the posts, I have a related content section that I usually use to complement the information of the tutorials. I also try to include references and links to the content I consult when creating the tutorials. Thanks :)

tushaar, 2017-05-08:
@anteph You could ask http://esp32.net to add your blog under reading and videos. I used that site to get started.

SarahAshraf, 2017-05-13:
can you post an example for BLE on esp-32 arduino?

mxwang101, 2017-05-14:
Yes, do need a BLE/WIFI for ESP32
,On 05/14/2017 05:22, SarahAshraf wrote:

can you post an example for BLE on esp-32 arduino?

anteph, 2017-05-15:
Hi! I haven't yet played with BLE sorry. But as soon as I have some content about it I will share here.

For WiFi I have this one bellow, which explains how to connect to a WiFi network. It's a very simple example and the method for connecting is similar to the one used in many of the examples of this repository.
https://techtutorialsx.com/2017/04/24/esp32-connecting-to-a-wifi-network/

Then, I have some more complex examples, built on top of the WiFi functionality, on how to publish and receive MQTT messages: 
https://techtutorialsx.com/2017/04/24/esp32-publishing-messages-to-mqtt-topic/
https://techtutorialsx.com/2017/04/24/esp32-subscribing-to-mqtt-topic/
https://techtutorialsx.com/2017/04/29/esp32-sending-json-messages-over-mqtt/

mxwang101, 2017-05-16:
good，waiting……, thx
On 05/16/2017 02:11, anteph wrote:

Hi! I haven't yet played with BLE sorry. But as soon as I have some content about it I will share here.

For WiFi I have this one bellow, which explains how to connect to a WiFi network. It's a very simple example and the method for connecting is similar to the one used in many of the examples of this repository.
https://techtutorialsx.com/2017/04/24/esp32-connecting-to-a-wifi-network/

Then, I have some more complex examples, built on top of the WiFi functionality, on how to publish and receive MQTT messages:
https://techtutorialsx.com/2017/04/24/esp32-publishing-messages-to-mqtt-topic/
https://techtutorialsx.com/2017/04/24/esp32-subscribing-to-mqtt-topic/
https://techtutorialsx.com/2017/04/29/esp32-sending-json-messages-over-mqtt/

skyler14, 2017-09-12:
Did you ever end up working BLE in arduino with the esp32, would love to see a tutorial for that.

anteph, 2017-09-13:
Hi! Unfortunately I haven't explored BLE yet, although I've been looking into Bluetooth Classic features. I have a couple of tutorials on those, although they are for ESP32 IDF. If they are of interest, let me know and I will share the links here.

Best regards,
Nuno Santos

skyler14, 2017-09-13:
Hi anteph, 

I would actually greatly appreciate that Bluetooth Classic tutorial for arduino. Thanks.

skyler14, 2017-09-13:
Hi anteph,

I would greatly appreciate those links.

Thanks,
Skyler

anteph, 2017-09-17:
Hi! Here are the links, hope they are helpful. As mentioned, they are for IDF, not Arduino:

- **Using the BTStack Library:** https://techtutorialsx.com/2017/07/08/esp32-bluetooth-using-the-btstack-library/
- **Advertising a service with spp:** https://techtutorialsx.com/2017/07/17/esp32-bluetooth-advertising-a-spp-service-with-sdp/
- **Receiving data with RFCOMM:** https://techtutorialsx.com/2017/07/09/esp32-bluetooth-rfcomm/
- **Finding a ESP32 BT device with Python:** https://techtutorialsx.com/2017/07/09/esp32-bluetooth-finding-the-device-with-python-and-btstack/

Would be awesome if someone could find a way of using the BTStack library on the Arduino core!

Best regards,
Nuno Santos

skyler14, 2017-09-17:
>    2017/07/08/esp32-bluetooth-using-the-btstack-library/
>    2017/07/17/esp32-bluetooth-advertising-a-spp-service-with-sdp/
>    2017/07/09/esp32-bluetooth-rfcomm/
>    2017/07/09/esp32-bluetooth-finding-the-device-with-python-and-btstack/

Hi anteph,
I'm familiar with arduino and have heard of micropython but not idf, can
you elaborate a bit. I thought it was a specific interface fit arduino
personally. Is it another developer environment

anteph, 2017-09-17:
Yes, it is another development environment. IDF (Espressif IoT Development Framework) is the official development framework for the ESP32, as you can see below:
https://github.com/espressif/esp-idf

IDF is programmed in C/C++ and the APIs are typically lower level than in the Arduino core, so it may be a little bit harder to understand if you don't have prior knowledge with these languages. Nonetheless, there are also plenty of tutorials and guides online which can help you getting started if you feel curious about it.

MicroPython is another way of programming the ESP32 (I also have a couple of posts on how to use it). Note that using MicroPython you write scripts that are interpreted rather than compiled, and thus the paradigm is different from the Arduino core. Nonetheless, it is equally easy to learn and much of the tutorials that are there for Python work for MicroPython.

You have plenty of other options for programming the ESP32:
- Javascript
- Lua
- PlatformIO
- Zerynth
- MongooseOS
(...)

You can check the website below which does a great job at compiling both tutorials and tools for programming the ESP32. It keeps getting regularly updated.
http://esp32.net/

kushiii-sys, 2019-09-06:
NORVI controllers comes with many communication features which makes it ideal for IOT solutions. You can order NORVI controller with a built-in WiFi module or W5500 Ethernet module and access your device online. Following are few products with higher IOT compatibility.

– ESP32 SOC based controllers for IOT applications

– NORVI IIOT Model with ESP32-WROOM32 as processor making a Industrial ESP32

– NORVI Arita with ESP8266 built-in

visit us www.norvi.lk
https://www.youtube.com/watch?time_continue=3&v=NVzUWdYsrF4

pedrominatel, 2022-05-17:
Hi @anteph, thanks for your community contribution!

We can add a new section on the [Arduino ESP32 Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/) and link your blog there.

Have you considered contributing to the GH project by adding some new examples?

Thanks!

anteph, 2022-05-23:
Hi @pedrominatel, you're welcome, I'm very happy to contribute with the tutorials  :)

Please feel free to add a link to my blog at the docs or in any place you think it could be useful for the community.

I've considered it but unfortunately, due to time limitations, I've never got involved directly in the project. Nonetheless, if you find some code in the tutorials that you think could serve as good example for the project, please also feel free to use it :) 

Thank you and to all the folks at Espressif for your awesome work!

pedrominatel, 2022-05-24:
Thanks @anteph!

I'll create this new section and then I'll close this issue.

Quando vieres à Braga, o fino é por minha conta!

PilnyTomas, 2023-06-15:
I'm closing this issue as the PR is merged.

closed 2023-06-15 by PilnyTomas