espressif/arduino-esp32 #365: Bug in Preferences library (or NVS).

Edzelf, 2017-05-09:
I found a bug in the Preferences library.  getString fails sometimes with error "getString(): nvs_get_str len fail".  The attached sketch shows the problem when "loop()" is called for the second time.
[prefbug.ino.txt](https://github.com/espressif/arduino-esp32/files/986105/prefbug.ino.txt)

Edzelf, 2017-05-11:
I tried to reproduce the bug in IDF, using the nvs-library.  I found the same problem.  The returned error is "ESP_ERR_NVS_NOT_FOUND" for most (but not all) keys in the namespace. See attached sourcecode.
Please help, for my application needs these function badly...
[testnvs.c.txt](https://github.com/espressif/arduino-esp32/files/993479/testnvs.c.txt)

me-no-dev, 2017-05-11:
if you found it not working in IDF, please submit the bug there. If it's fixed there, it will work here as well

igrr, 2017-05-12:
Thanks for the report, will fix this in IDF

igrr, 2017-05-12:
fixed in IDF in https://github.com/espressif/esp-idf/commit/bf01525fc154594d3b5eb7f57bfd83c9daf1359d

Edzelf, 2017-05-12:
Great!  Thanks for for your quick response.

me-no-dev, 2017-05-12:
Still in France. Will try to update Sunday or Monday

Edzelf, 2017-05-12:
Okay.  Now I know what the problem is, I have made a temporary workaround.

Edzelf, 2017-05-17:
@me-no-dev Please don't forget the fix.

me-no-dev, 2017-05-17:
@Edzelf I did update the libs :) You should be set

Edzelf, 2017-05-18:
Yes!  Works correct now.  Thanks.

closed 2017-05-18 by Edzelf

fenyvesi, 2017-09-07:
Seems to occur again in cases whwn I call it 100-150 times within 10-15 sec period.
 
D: nvsg 1
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x40080df2  PS      : 0x00060034  A0      : 0x80081104  A1      : 0x3ffc0bc0  
A2      : 0x000000ba  A3      : 0x00000001  A4      : 0x3ffc3c2c  A5      : 0x3ffcf9a0  
A6      : 0x00000000  A7      : 0x3ffcf9d0  A8      : 0x00000000  A9      : 0x3ffc0ba0  
A10     : 0x3f40223b  A11     : 0x3ffc36aa  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000003  A15     : 0x00060823  SAR     : 0x00000004  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000164d  LEND    : 0x40001667  LCOUNT  : 0xfffffffd  

Backtrace: 0x40080df2:0x3ffc0bc0 0x40081104:0x3ffc0be0 0x40082205:0x3ffc0c00

CPU halted.