espressif/arduino-esp32 #368: WiFiClientSecure and Google App Script

bestpika, 2017-05-10:
I can't get any response.
```cpp
WiFiClientSecure client;
...
if(client.connect("script.google.com", 443)) {
    client.println("GET /macros/s/.../exec HTTP/1.1");
    client.println("Host: script.google.com");
    client.println("Content-Type: application/x-www-form-urlencoded");
    client.println("Content-Length: 3");
    client.println();
    client.println("a=a");
    client.print("\r\n\r\n");
    Serial.print("Waiting for response ");
    while (!client.available()) { // loop forever
        delay(50);
        Serial.print(".");
    }
    while (client.available()) {
        char c = client.read();
        Serial.write(c);
    }
    client.stop();
}
```

copercini, 2017-05-10:
I fixed this possible infinite loop here: https://github.com/espressif/arduino-esp32/pull/347/files#diff-78da625bea886a01b784f9d1563acf3aL90

But it's not merged yet

electronicsguy, 2017-05-11:
@bestpika you won't get a direct response from google apps script since it uses redirection. Use this: [HTTPSRedirect](https://github.com/electronicsguy/ESP8266/tree/master/HTTPSRedirect). 

It works well on esp8266. I'll test it on ESP-32 and update. In case you test it and get any errors, please post them.

bestpika, 2017-05-11:
@electronicsguy I can't get any response, so HTTPSRedirect not work.

copercini, 2017-05-11:
@bestpika please enable Core Debug Level as verbose in your arduino IDE and post here

electronicsguy, 2017-05-11:
@bestpika that probably indicates an error in the google apps scripts permissions. What settings is your we app published with?

bestpika, 2017-05-12:
@electronicsguy
[image]

electronicsguy, 2017-05-12:
@bestpika   You can use a service like [hurl](hurl.it) to test the webapp externally. Please see if that gives any response and errors, if any. Could you post the webapp URL here?

bestpika, 2017-05-12:
If no post data, I can get response, but result is wrong.
If I send post data, I can get any response.

me-no-dev, 2017-05-12:
What I can tell is that ```a=a``` is not url encoded, therefore the request is wrong. That value should be ```a%3Da```

me-no-dev, 2017-05-12:
but that could also be caused by wrong account data

bestpika, 2017-05-12:
If app script no return, then not redirect.
I will change send data and try.

bestpika, 2017-05-12:
I rewrite http request, then It is working okay.
```cpp
void doSendThis() {
  Serial.println("Send...");
  if (client.connect(HOST, 443)) {
    String postData(String("gssid=") + GSSID + "&sheet=" + SHEET + "&value=" + "0|1|2");
    Serial.println(postData);
    client.println("POST " + String(PATH) + " HTTP/1.1");
    client.println("Host: " + String(HOST));
    client.println("Content-Type: application/x-www-form-urlencoded");
    client.println("Content-Length: " + String(postData.length()));
    client.println();
    client.println(postData);
    client.print("\r\n\r\n");
    Serial.println("Response...");
    while (!client.available()) {
      delay(100);
      Serial.print(".");
    }
    Serial.println();
    Serial.println("-----------");
    while (client.available()) {
      Serial.write(client.read());
    }
    Serial.println();
    Serial.println("-----------");
  }
  client.stop();
}
```

closed 2017-05-12 by bestpika

swetapriyadarshi, 2018-03-14:
@bestpika 
Hey i know this post is closed but still can u post the complete method of datalogging in google sheet as i am not able to do it.
Thanks

chandan2002x, 2018-06-11:
At last uploaded to Spreadsheet with esp-idf. But loosing 76 byte heap at every transaction. Any clue?

Voha888, 2019-10-22:
> I rewrite http request, then It is working okay.
> 
> ```c++
> void doSendThis() {
>   Serial.println("Send...");
>   if (client.connect(HOST, 443)) {
>     String postData(String("gssid=") + GSSID + "&sheet=" + SHEET + "&value=" + "0|1|2");
>     Serial.println(postData);
>     client.println("POST " + String(PATH) + " HTTP/1.1");
>     client.println("Host: " + String(HOST));
>     client.println("Content-Type: application/x-www-form-urlencoded");
>     client.println("Content-Length: " + String(postData.length()));
>     client.println();
>     client.println(postData);
>     client.print("\r\n\r\n");
>     Serial.println("Response...");
>     while (!client.available()) {
>       delay(100);
>       Serial.print(".");
>     }
>     Serial.println();
>     Serial.println("-----------");
>     while (client.available()) {
>       Serial.write(client.read());
>     }
>     Serial.println();
>     Serial.println("-----------");
>   }
>   client.stop();
> }
> ```
> 
> ```
> HTTP/1.1 200 OK
> Content-Type: text/html; charset=utf-8
> Cache-Control: no-cache, no-store, max-age=0, must-revalidate
> Pragma: no-cache
> Expires: Mon, 01 Jan 1990 00:00:00 GMT
> Date: Fri, 12 May 2017 06:45:28 GMT
> P3P: CP="This is not a P3P policy! See https://support.google.com/accounts/answer/151657?hl=en for more info."
> P3P: CP="This is not a P3P policy! See https://support.google.com/accounts/answer/151657?hl=en for more info."
> X-Content-Type-Options: nosniff
> X-XSS-Protection: 1; mode=block
> Server: GSE
> Set-Cookie: ...
> Set-Cookie: ...
> Alt-Svc: quic=":443"; ma=2592000; v="37,36,35"
> Accept-Ranges: none
> Vary: Accept-Encoding
> Transfer-Encoding: chunked
> ```
> 
> [image]

Hello. What is the value in your PATH string?

Amrutha-bv, 2020-05-23:
> Hello. What is the value in your PATH string?

Were you able to get the value for PATH? Please let me know, HttpsRedirect didnot work for me

bestpika, 2020-05-23:
The HOST and PATH just GAS's execute URL.

Amrutha-bv, 2020-05-25:
I did try this.

My google script ID recognizes "Name" and pastes it in that respective column.

``
const char* host = "script.google.com";
String url = String("/macros/s/") + GScriptId + "/exec?cal";
payload = "Name=hi";
client.println("POST` " + String(url) + " HTTP/1.1");
    client.println("Host: " + String(host));
    client.println("Content-Type: application/x-www-form-urlencoded");
    client.println("Content-Length: " + String(payload.length()));
    client.println();
    client.println(payload);
    client.print("\r\n\r\n");`

I am able to read string rceived from host. Data is not being received at spreadsheet. What am I missing here?

Nimmi147, 2020-07-05:
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include "DHT.h"
#define DHTPIN D1
#define DHTTYPE DHT11 
 
const char* ssid     = "ningu";
const char* password = "14071996";
const char* host = "script.google.com"; 
const char* fingerprint = "‎‎CD 3E 63 37 0D 07 56 7A 9B 10 6B 7A 3D 64 57 9B 1E 4B B6 D5";
String url;

DHT dht(DHTPIN, DHTTYPE);

void setup() {
  Serial.begin(115200);
  delay(100);
  dht.begin();
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password); 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.print("Netmask: ");
  Serial.println(WiFi.subnetMask());
  Serial.print("Gateway: ");
  Serial.println(WiFi.gatewayIP());
}

 
void loop() {
 
  Serial.print("connecting to ");
  Serial.println(host);
 
  WiFiClientSecure client;

  const int httpsPort = 443;
  if (!client.connect(host, httpsPort)) {
    Serial.println("connection failed");
    return;
  }

 if (client.verify(fingerprint, host)) {
  Serial.println("certificate matches");
  } else {
  Serial.println("certificate doesn't match");
  }

  float h = dht.readHumidity();
  float t = dht.readTemperature();
  if (isnan(h) || isnan(t)) {
     Serial.println("Failed to read from DHT sensor!");
     return;
  }
  
  url = "/macros/s/AKfycbzBr2x17f5TMmWrV_rs3SF8cyLYlRG52nfwzAoAluaDD-jAsDI/exec?func=addData&val="+ String(t);
  Serial.print("Requesting URL: ");
  Serial.println(url);
  
  client.print(String("GET ") + url + " HTTP/1.1\r\n" +
               "Host: " + host + "\r\n" + 
               "Connection: close\r\n\r\n");
  delay(500);
  String section="header";
  while(client.available()){
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }
  Serial.println();
  Serial.println("closing connection");
  delay(3000);
}
[image]

It showing an error called connecting to script.google.com connection failed please help me where i did mistake.