espressif/arduino-esp32 #369: Neopixel.h

Buffalchill, 2017-05-10:
Hi,
Can someone tell me please if the Neopixel.h library works with the ESP32 ?
I tried the Strandtest example but it does not work on my DEF board.

Regards
Stefan

gardi-innovation, 2017-05-10:
Hi,
I've used [this](https://github.com/Makuna/NeoPixelBus) library with success on the ESP8266. 
I'm currently in the progress of porting my code to the ESP32, but am putting focus on other tasks at the moment.
Maybe you can try it out and tell me if it works?

Buffalchill, 2017-05-11:
I tried it with ESP8266 it works but my pixel animations get stuck because of wifi stuff...
I also tried it on ESP32 but it does not work there yet. "platform currently not supported"

forthlightning, 2017-05-12:
https://www.esp32.com/viewtopic.php?f=18&t=595&p=5258&hilit=ws2812#p5258

I use this and it works fine. Neil is a G.

Buffalchill, 2017-05-13:
@forthlightning but this is not a Arduino library right ? how do i use this in Arduino on the ESP32 ?

rmhomecouk, 2017-05-14:
@Buffalchill the latest neopixelbus supports ESP32 - Have you grabbed the latest code? It's working for me.

Buffalchill, 2017-05-14:
YA thank you ! I grabbed the latest now it works ty

closed 2017-05-19 by me-no-dev