espressif/arduino-esp32 #371: Cannot turn off serial debug message

saknarak, 2017-05-12:
```
Serial.begin(115200);
Serial.setDebugOutput(0);
```
Still have debug message out to serial

For example
```
E (442432) BT: BLE HCI(id=62) event = 0x02)
E (442432) BT: BLE HCI(id=62) event = 0x02)
E (442448) BT: BLE HCI(id=62) event = 0x02)
...
```

How I totally turn off serial debug message?

my code based on example from ` https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/gatt_client`

me-no-dev, 2017-05-12:
aaahhh I might have an idea. Can you share some code so I can try to reproduce it? Or will it work with the example on it's own?

saknarak, 2017-05-12:
Yes sure, here is my code
```
// Based from the IDF GATT Client example - https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/gatt_client

#include <stdint.h>
#include <string.h>
#include <stdbool.h>
#include <stdio.h>
#include "controller.h"

#include "bt.h"
#include "bt_trace.h"
#include "bt_types.h"
#include "btm_api.h"
#include "bta_api.h"
#include "bta_gatt_api.h"
#include "esp_gap_ble_api.h"
#include "esp_gattc_api.h"
#include "esp_gatt_defs.h"
#include "esp_bt_main.h"

void gattc_client_test(void);

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(0);
  // just ble scannner
  if (btStart()) {
    gattc_client_test();
  }
}

void loop() {
  // do nothing
}

void bleDumpResultHex(esp_ble_gap_cb_param_t *scan_result) {
  Serial.printf("FOUND: %02X%02X%02X%02X%02X%02X ", scan_result->scan_rst.bda[0], scan_result->scan_rst.bda[1], scan_result->scan_rst.bda[2], scan_result->scan_rst.bda[3], scan_result->scan_rst.bda[4], scan_result->scan_rst.bda[5]);
  byte totalLen = scan_result->scan_rst.adv_data_len + scan_result->scan_rst.scan_rsp_len;
  for (byte i = 0; i < totalLen; i++) {
    Serial.printf("%02X", scan_result->scan_rst.ble_adv[i]);
  }
  Serial.write("\r\n");
}

static esp_ble_scan_params_t ble_scan_params = {
  .scan_type              = BLE_SCAN_TYPE_ACTIVE,
  .own_addr_type          = BLE_ADDR_TYPE_PUBLIC,
  .scan_filter_policy     = BLE_SCAN_FILTER_ALLOW_ALL,
  .scan_interval          = 0x50,
  .scan_window            = 0x30
};

static void esp_gap_cb(esp_gap_ble_cb_event_t event, esp_ble_gap_cb_param_t *param) {
  uint8_t *adv_name = NULL;
  uint8_t adv_name_len = 0;
  uint8_t *adv_data = NULL;
  uint8_t adv_data_len = 0;
  
  switch (event) {
  case ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT: {
    //the unit of the duration is second
    uint32_t duration = 300;
    esp_ble_gap_start_scanning(duration);
    break;
  }
  case ESP_GAP_BLE_SCAN_RESULT_EVT: {
    esp_ble_gap_cb_param_t *scan_result = (esp_ble_gap_cb_param_t *)param;
    switch (scan_result->scan_rst.search_evt) {
    case ESP_GAP_SEARCH_INQ_RES_EVT: {
      bleDumpResultHex(scan_result);
      break;
    }
    case ESP_GAP_SEARCH_INQ_CMPL_EVT: {
      // add code to start another scan
      Serial.println("scan completed, restarting ...");
      esp_ble_gap_set_scan_params(&ble_scan_params);
//    esp_ble_gap_start_scanning(300);
      break;
    }
    default:
      Serial.printf("UNHANDLE ESP_GAP_SEARCH_EVT: %d\n", scan_result->scan_rst.search_evt);
      break;
    }
    break;
  }
  default:
    Serial.printf("UNHANDLE ESP_GAP_BLE_SCAN_EVT: %d\n", event);
    break;
  }
}

void ble_client_appRegister(void)
{
  esp_err_t status;

  Serial.printf("register callback\n");

  //register the scan callback function to the gap moudule
  if ((status = esp_ble_gap_register_callback(esp_gap_cb)) != ESP_OK) {
    Serial.printf("ERROR: gap register error, error code = %x\n", status);
    return;
  } else {
    esp_ble_gap_set_scan_params(&ble_scan_params);
  }
}

void gattc_client_test(void)
{
  esp_bluedroid_init();
  esp_bluedroid_enable();
  ble_client_appRegister();
}
```

and this is the result
```
FOUND: F2CCF95A63CA 0201061AFF4C00021574278BDAB64445208F0C720EAF059935271150E5C5020A040816F0FF64271150E511094D696E69426561636F6E5F3230373039
E (52754) BT: BLE HCI(id=62) event = 0x02)
E (52754) BT: BLE HCI(id=62) event = 0x02)
FOUND: F4B85EE508B2 0201061AFF4C000215FDA50693A4E24FB1AFCFC6EB0764782500050000C5020A000816F0FF640005000012095379536974735F32343730360000000000
E (52763) BT: BLE HCI(id=62) event = 0x02)
E (53064) BT: BLE HCI(id=62) event = 0x02)
FOUND: 3015277F3A50 1EFF0600010920007A7CE094CCE9B79ABA158D35C7F9D23194AD753F27F2C2
E (53352) BT: BLE HCI(id=62) event = 0x02)
E (53352) BT: BLE HCI(id=62) event = 0x02)
FOUND: D42202000223 0201061AFF4C00021574278BDAB64445208F0C720EAF05993500010223C5020A000816F0FF640001022312095379736974735F30303534370000000000
E (53657) BT: BLE HCI(id=62) event = 0x02)
E (53657) BT: BLE HCI(id=62) event = 0x02)
FOUND: F4B85EE508B2 0201061AFF4C000215FDA50693A4E24FB1AFCFC6EB0764782500050000C5020A000816F0FF640005000012095379536974735F32343730360000000000
E (53790) BT: BLE HCI(id=62) event = 0x02)
E (53790) BT: BLE HCI(id=62) event = 0x02)
FOUND: D42202000005 0201061AFF4C0002156E732E636F2E74682D69626561636F6E00010005C5020A000816F0FF640001000512095379536974735F30303030350000000000
E (53964) BT: BLE HCI(id=62) event = 0x02)
E (54091) BT: BLE HCI(id=62) event = 0x02)
E (54091) BT: BLE HCI(id=62) event = 0x02)
FOUND: D42202000005 0201061AFF4C0002156E732E636F2E74682D69626561636F6E00010005C5020A000816F0FF640001000512095379536974735F30303030350000000000
E (54400) BT: BLE HCI(id=62) event = 0x02)
E (54400) BT: BLE HCI(id=62) event = 0x02)
FOUND: D42202000005 0201061AFF4C0002156E732E636F2E74682D69626561636F6E00010005C5020A000816F0FF640001000512095379536974735F30303030350000000000
E (54564) BT: BLE HCI(id=62) event = 0x02)
E (54703) BT: BLE HCI(id=62) event = 0x02)
E (54703) BT: BLE HCI(id=62) event = 0x02)
FOUND: D42202000005 0201061AFF4C0002156E732E636F2E74682D69626561636F6E00010005C5020A000816F0FF640001000512095379536974735F30303030350000000000
E (54864) BT: BLE HCI(id=62) event = 0x02)
E (55006) BT: BLE HCI(id=62) event = 0x02)
E (55006) BT: BLE HCI(id=62) event = 0x02)
FOUND: D42202000005 0201061AFF4C0002156E732E636F2E74682D69626561636F6E00010005C5020A000816F0FF640001000512095379536974735F30303030350000000000
E (55144) BT: BLE HCI(id=62) event = 0x02)
E (55144) BT: BLE HCI(id=62) event = 0x02)
FOUND: EFE309CBC23B 0201061109536D6172742048756D69676164676574
```

How to suppress debug message of BLE

me-no-dev, 2017-05-12:
You probably can't, but I will recompile the libs and get rid of those. Last time I updated the libs, I enabled HCI and that is what is spewing those errors.

me-no-dev, 2017-05-19:
can you see if the latest updates fixed this?

saknarak, 2017-05-19:
Yes, now is no more debug message.
Thank you very much

closed 2017-05-19 by me-no-dev