espressif/arduino-esp32 #372: can not run 2 encoders

cnc4less, 2017-05-12:
this the code i made for two encoders, the problem is that  both encoders
will have the same pins and will share the same results,  for example

if setup then the fEnc will be working and pEnc will just share the same value
pEnc.begin();
fEnc.begin();

if setup then the pEnc will be working and fEnc will just share the same value
fEnc.begin();
pEnc.begin();

maybe something wrong  with the attacfuction with esp32 !!! please advise.

espEncoder class 
void espEncoder::begin() {
  pinMode(encoderPinA, INPUT_PULLUP);
  pinMode(encoderPinB, INPUT_PULLUP);
  attachInterrupt(encoderPinA, read_encoder, CHANGE);
  attachInterrupt(encoderPinB, read_encoder, CHANGE);

------------------------------------------------------
#include "Arduino.h"
#include "espEncoder.h"

espEncoder freqEnc(4, 5);
espEncoder pwmEnc(16, 17);

volatile int NewValueFreq, OldValueFreq = 0;
volatile int NewValuePwm, OldValuePwm = 0;

void setup() {

  Serial.begin(115200);
  delay(1000);
  pEnc.begin();
  fEnc.begin();
}

void loop() {
  NewValueFreq = freqEnc.Value();
  if (OldValueFreq != NewValueFreq) {
    Serial.print("Freq  ? = ");
    Serial.println(NewValueFreq);
    OldValueFreq = NewValueFreq;
  }

  delay(50);

  NewValuePwm = pwmEnc.Value();
  if (OldValuePwm != NewValuePwm) {
  Serial.print("Pwm  ? = ");
  Serial.println(NewValuePwm);
  OldValuePwm = NewValuePwm;
  }
}

me-no-dev, 2017-05-12:
Maybe post more of the class source? Where is the constructor? Class properties?

cnc4less, 2017-05-12:
/*
   espEncoder.cpp

    Created on: Aug 9, 2014
        Author: Konstantin Gredeskoul

 

#include "espEncoder.h"
#include "Arduino.h"

volatile int16_t encoderPoss = 0;
uint8_t encoderPinAA;
uint8_t encoderPinBB;
uint8_t encoderPinButton;
uint8_t old_AB = 0;

// grey code
// http://hades.mech.northwestern.edu/index.php/Rotary_Encoder
// also read up on 'Understanding Quadrature Encoded Signals'
// https://www.pjrc.com/teensy/td_libs_Encoder.html
int8_t enc_states[] = { 0, -1, 1, 0, 1, 0, 0, -1, -1, 0, 0, 1, 0, 1, -1, 0 };
//static int16_t lastencoder = 0;

}
* /

void  read_encoder() {

  old_AB <<= 2;
  old_AB |= ((digitalRead(encoderPinBB)) ? (1 << 1) : 0)
            | ((digitalRead(encoderPinAA)) ? (1 << 0) : 0);
  encoderPoss += (enc_states[(old_AB & 0x0F)]);
}

espEncoder::espEncoder(uint8_t rotaryPinA, uint8_t rotaryPinB)

{

  encoderPinA = rotaryPinA;
  encoderPinB = rotaryPinB;

  encoderPinAA = rotaryPinA;
  encoderPinBB = rotaryPinB;

}

void espEncoder::begin() {

  pinMode(encoderPinA, INPUT_PULLUP);
  pinMode(encoderPinB, INPUT_PULLUP);

  attachInterrupt(encoderPinA, read_encoder, CHANGE);
  // delay(50);

  attachInterrupt(encoderPinB, read_encoder, CHANGE);
}

int16_t espEncoder::Value() {

  return  encoderPoss /  4;

}

cnc4less, 2017-05-12:
/*
 * espEncoder.h
 *
 *  Created on: Apr 24, 2017
 *      Author: goodness
 */

#ifndef ESPENCODER_H_
#define ESPENCODER_H_

/*
 * espEncoder
 *
 *  Created on: Aug 9, 2014
 *      Author: Konstantin Gredeskoul
 *
 *  (c) 2014 All rights reserved.  Please see LICENSE.
 *
 *  Example of such Rotary Encoder is this one:
 *  https://www.adafruit.com/products/377
 *
 */

/*
 #if defined(ARDUINO) && ARDUINO >= 100
 #include <Arduino.h>
 #else
 #include //<WProgram.h>
 #include <pins_arduino.h>
 #endif
 */

#include <Arduino.h>
//#include "Encoder.h"
//#include <pins_arduino.h>

#define DEBOUNCE_DELAY 		30

// once you press the button, no other events to the button matter
// until this time has passed.
#define BUTTON_PRESS_DELAY 300

class espEncoder {
public:
	// int16_t encoderPos = 0;
	 espEncoder(uint8_t rotaryPinA, uint8_t rotaryPinB);
	// void read_encoder();
	void begin();
	bool buttonClicked();
	int16_t Value(); 
 
private:
//int16_t Value();
  int16_t encoderPos = 0;
    uint8_t encoderPinA;
    uint8_t encoderPinB;
	int _buttonPin;
	bool _hasNotReadRotary;
	unsigned long int _lastButtonPressedAt;
	signed long int _lastRotaryValue;
};

#endif /* ESPENCODER_H_ */

cnc4less, 2017-05-12:
serial output

Freq  ? = -1
Pwm  ? = 0
Freq  ? = 0
Pwm  ? = -1
Freq  ? = -1
Pwm  ? = -2
Freq  ? = -2
Pwm  ? = -3
Freq  ? = -3
Pwm  ? = -4
Freq  ? = -4
Pwm  ? = -5
Freq  ? = -5

me-no-dev, 2017-05-12:
Those are global so maybe that is your problem
```cpp
volatile int16_t encoderPoss = 0;
uint8_t encoderPinAA;
uint8_t encoderPinBB;
uint8_t encoderPinButton;
uint8_t old_AB = 0;
```

cnc4less, 2017-05-12:
i will check  later today I really hope it works, Than you for your kind support.

me-no-dev, 2017-05-19:
it surely is somewhere in the code :) there is no other reason. Sadly I do not have the time currently to get two encoders and go through the code :(

closed 2018-04-19 by everslick