espressif/arduino-esp32 #373: SimpleBLE.h example problem

MarconiLab, 2017-05-12:
Hi, I am running SimpleBLEDevice.h code on a NodeMCU ESP-32S board.

When trying to do ble.begin("ESP32"); on the setup the code gets stuck. Digging a little bit the code gets stuck inside ble_send_cmd() at the end

while(_vhci_host_command_running);
int res = _vhci_host_command_result;

It never gets out from the while.
Thanks

me-no-dev, 2017-05-12:
I'll give it a go hopefully sunday or monday and will reply back. It should work.

ESAlexhi, 2017-05-15:
Yes, I have the same problem

sarim987, 2017-05-17:
I am also experiencing this problem. Thanks

me-no-dev, 2017-05-19:
I did a good couple of hours digging through this. And there is no apparent reason why it would not work, but it does not. Callbacks are not called and the code can not progress. The included example in IDF has not changed either, so there are no clues to what has changed or what can be done. Maybe good time to remove that SimpleBLE example and be done for now.
My test GATT service is working, so not all is broken.

ESAlexhi, 2017-05-19:
Where can I download and try  your  test GATT service?

cnxsoft, 2017-05-19:
Just for reference, I tested SimpleBLE around  May 6th on ESP32-Bit/ESP32-T, and it worked at the time.

GenbandCloud, 2017-05-20:
I also had this problem with today's code. I did a 
    git checkout `git rev-list -1 --before="May 05 2017" master`
in esp32 repository and now when I run the SimpleBleDevice example I actually see the "Press the button to change the device name" prompt. I'm using the latest Arduino IDE.

I see multiple changes to BT related files in esp32 on May 6. 

hope this helps.

cnxsoft, 2017-05-20:
That's what I used: 
ESP32 SDK: v2.0-rc1-443-g3cad00f-dirty
BTDM CONTROLLER VERSION: 010101

The OP started to have the problem on May 11th, so something changed between May 6th and May 11th.

closed 2017-05-22 by me-no-dev

MarconiLab, 2017-05-22:
With the new code I have this

ESP32 SDK: v2.0-rc1-761-g65acd99c
BTDM CONTROLLER VERSION: 010101
btip start
copy .data from 4000d890 to 3ffae6e0, len 00001830
set .bss 0x0 from 3ffb8000 to 3ffbff70, len 00007f70
BTDM ROM VERSION 0101
BD_ADDR: 30:AE:A4:01:CF:62
NVDS MAGIC FAILED
RF Init OK with coex
Enable Classic BT
Enable Low Energy
E (4860) BT: Startup BTU

E (4861) BT: L2CA_RegisterFixedChannel()  CID: 0x0004, 0x3ffdc0c0
E (4861) BT: GATTS_CreateService:  handle of service handle1
E (4863) BT: L2CA_RegisterFixedChannel()  CID: 0x0006, 0x3ffdc0a0
E (4869) BT: L2CA_RegisterFixedChannel()  CID: 0x0007, 0x3ffdc0a0
E (4887) BT: #####################################num_lm_ble_bufs = 10
E (4889) BT: bta_dm_co_ble_load_local_keys: func not ported

Press the button to change the device name
E (4902) BT: ####################adv_type = 3

But pressing the button nothing changes. I can see on my phone ESP32 SimpleBLE but not pairing

me-no-dev, 2017-05-22:
Running the SimpleBLE example from Arduino IDE gives me this result and it's changing it's name:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
ESP32 SDK: v2.0-rc1-803-g1e0710f1
BTDM CONTROLLER VERSION: 010101
btip start
copy .data from 4000d890 to 3ffae6e0, len 00001830
set .bss 0x0 from 3ffb8000 to 3ffbff70, len 00007f70
BTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:00:7F:26
NVDS MAGIC FAILED
RF Init OK with coex
Enable Classic BT
Enable Low Energy
Press the button to change the device name
BLE32 at: 21
BLE32 at: 23
BLE32 at: 23
BLE32 at: 24
BLE32 at: 24
BLE32 at: 24
BLE32 at: 25
BLE32 at: 25
BLE32 at: 26
BLE32 at: 26
BLE32 at: 26
BLE32 at: 27
BLE32 at: 27
```

ESAlexhi, 2017-05-23:
Yes! Its work ok. Snx ,   me-no-dev!