espressif/arduino-esp32 #374: Missing xtensa-esp32-elf when install on Raspberry Pi

whizzCoder, 2017-05-12:
When I followed the instruction for Linux to install on Raspberry Pi, the xtensa-esp32-elf folder is missing. Is it supported on Raspberry Pi?

Thanks.

me-no-dev, 2017-05-12:
Not officially yet, but you can build the toolchain on it and use it.

whizzCoder, 2017-05-12:
Thanks for the advise. 
is there any instructions somewhere? :)

me-no-dev, 2017-05-12:
I think on the esp-idf repo readme/docs. On the phone currently so its hard to find :)

whizzCoder, 2017-05-12:
Thanks ... I'm still searching for the doc. :)

Hi everyone else, if you happen to know any place that I can download the toolchain for raspberry pi 3, or instruction on how to build the toolchain, please let me know. Million thanks. :)

tobozo, 2017-05-12:
[Pre built binaries for rpi3](http://www.neilkolban.com/esp32/downloads/xtensa-esp32-elf.tar.gz) but it's not recent, also [this video](https://youtu.be/jt0aaMQD1WI)

me-no-dev, 2017-05-12:
https://github.com/espressif/crosstool-NG

whizzCoder, 2017-05-13:
Thanks for all the help.
I've started to compile the toolchain.
Almost 2h passed, likely many more to go. :)

whizzCoder, 2017-05-13:
The toolchain build took 170mins to complete.
I'm able to compile and upload a blinking LED sketch to my ESP32 dev board successfully.
Thank you all for your help.
Happy weekend!

closed 2017-05-19 by me-no-dev

pearsilva, 2018-10-16:
> The toolchain build took 170mins to complete.
> I'm able to compile and upload a blinking LED sketch to my ESP32 dev board successfully.
> Thank you all for your help.
> Happy weekend!

Could you help me with this?
I'm starting my knowledge on ESP32 using the Raspberry Pi.

Thanks for your attention.

urbanze, 2019-04-02:
Hi guys, I trying to do this but my knowledge in linux is not the best. When compile IDF, terminal return something similar

`make all -j4
/home/pi/mcu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/pi/mcu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
expr: syntax error
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
Toolchain path: /home/pi/mcu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu`

What I need to do? I not understand "crosstoll-ng"...

beegee-tokyo, 2019-04-03:
ESP32 toolchain is not available readily for the Raspberry.
Some time ago someone compiled the toolchain for the RPi, but I do not know if it still works:
[Building ESP32 toolchain on ARM (Armbian Raspbian)](https://desire.giesecke.tk/index.php/2018/01/30/building-esp32-toolchain-on-arm-armbian-raspbian/)

tobozo, 2019-04-03:
@beegee-tokyo I followed the [video](https://youtu.be/jt0aaMQD1WI) tutorial a couple of weeks ago, and it worked just fine.

urbanze, 2019-04-06:
Well, sucess! I use crosstool-NG to recompile toolchain for arm (raspberry), worked perfecly in most recent master branch of IDF and last toolchain. Followed tutorial:https://docs.espressif.com/projects/esp-idf/en/latest/get-started/linux-setup-scratch.html

syswombat, 2019-07-23:
Hello
on a Raspberry Pi 4 with Buster is there a easy way to solve the problem!

Tool xtensa-esp32-elf-gcc is not available for your operating system.
java.lang.RuntimeException: java.lang.Exception: Tool xtensa-esp32-elf-gcc is not available for your operating system.
at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Tool xtensa-esp32-elf-gcc is not available for your operating system.
at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:85)
at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
... 1 more