espressif/arduino-esp32 #378: LEDC can't set duty cycle to 100%

maxpromer, 2017-05-17:
Hello!, on NodeMCU-32S i test LEDC via my code.
```
int value = 2500;

void setup() {
  ledcSetup(0, 5000, 13);
  ledcAttachPin(LED_BUILTIN, 0);
  pinMode(0, INPUT);
  ledcWrite(0, value);
}

void loop() {
  if (digitalRead(0) == LOW) {
    while(digitalRead(0) == LOW) delay(10);
    ledcWrite(0, value = (value >= 5000) ? 0 : value + 2500);
  }
} 
```
I enter button to max duty cycle, my oscilloscope show 61%. How to fix this ?

[image]

maxpromer, 2017-05-17:
OK. i edit code to 

```
int value = 8100;

void setup() {
  ledcSetup(0, 5000, 13);
  ledcAttachPin(LED_BUILTIN, 0);
  pinMode(0, INPUT);
  ledcWrite(0, value);
}

void loop() {

}
```

8100 from 5000 / 61% * 100 = 8196 i try set `value` to 8196 my oscilloscope show  ?% (about 0%) then i edit value to 8100 my oscilloscope show  99%, How set to duty cycle to 100% ?

igrr, 2017-05-17:
Guessing 2^13-1=8191

maxpromer, 2017-05-17:
in example `LEDCSoftwareFade` use `LEDC_BASE_FREQ` to calculate duty cycle, Why ?

```
  uint32_t duty = (LEDC_BASE_FREQ / valueMax) * min(value, valueMax);

  // write duty to LEDC
  ledcWrite(channel, duty);
```

in example set `LEDC_BASE_FREQ` to 5000 and `valueMax` to 255 if input `value = 255` variable
  `duty` not equal to 100% duty cycle. (5000 / 255 * 255 = 5000 and not equal to 100% duty cycle, 8191 = 100%)

closed 2017-05-19 by me-no-dev