espressif/arduino-esp32 #380: Sleep modes

yomasa, 2017-05-18:
Can someone point me to any code examples of Sleep modes with rtc?

copercini, 2017-08-27:
There are some examples of Deep Sleep with RTC support: https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/DeepSleep

closed 2017-09-29 by copercini

beats0126, 2018-06-27:
@copercini may i ask for the deep sleep problem, i using the example TouchWakeUp to do the deep sleep. i set the pin 15 as pin to call esp32 wakeup. but the problem i meet is it keep wakeup after going sleep. i am a beginner to this field. for checking the problem, i removed all wire from the board and only connect 1 f2f to my esp32 (espduino32), I also found if my finger touch pin 15 it also causing esp32 wake up. is it the logic to wakeup the esp32 when it receive a small power from the pin we selected?

beats0126, 2018-06-27:
i tried with 2 espduino32 and 1 lolin32 board. the result is same. may i have a help for this issue? because i going to do using mpr121 touch sensor to wakeup my esp32. but the problem is it keep wakeup after sleep.