espressif/arduino-esp32 #382: No module named requests

architmuchhal12, 2017-05-18:
Hello. Using Windows 10 here. Followed the instructions on getting set up and receiving the following when trying to upload a blinking light sketch:

```
> $ python get.py
> Traceback (most recent call last):
>   File "get.py", line 25, in <module>
>     import requests
> ImportError: No module named requests
```

Followed the existing solutions by installing pip. Still did not work out.
Any suggestions?
Thanks for helping out!

me-no-dev, 2017-05-18:
for windows there is get.exe :)

architmuchhal12, 2017-05-18:
So, after using the Git GUI to clone the directory, I did as per the process given in the tutorial by double-clicking the "get.exe".  (https://github.com/espressif/arduino-esp32/blob/master/doc/windows.md)
Still,I could not view the "xtensa-esp32-elf" folder as shown in the tutorial.

[image]

architmuchhal12, 2017-05-22:
So, I did this same process and was able to extract _xtensa-esp32-elf_ folder. I had some issues with my computer so had to format. Now, again after repeating the same process, I am unable to extract the folder. After double clicking the get.exe file, the command prompt window opens and it gets closed automatically in a second without the extraction of the _xtensa-esp32-elf_ folder.
Please correct me if I am doing anything wrong.
Any help will be appreciated.
Thanks!

me-no-dev, 2017-05-23:
but I clearly see the xtensa-esp32-elf folder in your screenshot. What are you expecting to happen?

architmuchhal12, 2017-05-23:
That is the screenshot of the tutorial. It is not my computer. Anyway, I tried this process on my home laptop, and it worked. Not sure what is the issue.

me-no-dev, 2017-05-23:
You can execute get.exe through command prompt and see what is going wrong.

architmuchhal12, 2017-05-23:
[image]

This is the error which I am getting. I assume it is probably because of the security access to the URL from which the _xtensa_ folder is downloaded.

me-no-dev, 2017-05-23:
You can manually download [The Toolchain](https://dl.espressif.com/dl/xtensa-esp32-elf-win32-1.22.0-61-gab8375a-5.2.0-2.zip) and [EspTool](https://dl.espressif.com/dl/esptool-96698a3-windows.zip) to ```tools\dist\``` and then rerun get.exe which will unpack them.

architmuchhal12, 2017-05-23:
Yes! It worked!
Thanks!

closed 2017-05-23 by architmuchhal12

armandojaleo, 2018-03-04:
Works for me too.
Thanks.

blacxcode, 2019-07-05:
> 
> 
> @samjulien and @MrAPierce, need help!
> Hello. Using Windows 10 here. Followed the instructions on getting set up and receiving the following when trying to upload a blinking light sketch:
> (Using Arduino 1.8.2 and Python 2.7.13)
> 
> ```
> > $ python get.py
> > Traceback (most recent call last):
> >   File "get.py", line 25, in <module>
> >     import requests
> > ImportError: No module named requests
> ```
> 
> Followed the existing solutions by installing pip. Still did not work out.
> Any suggestions?
> Thanks for helping out!

you don't have installed "requests" lib in python.

open MINGW : 
$ python -m pip install requests

and then after finish installed 

$ python get.py