espressif/arduino-esp32 #383: BLE will be fully supported?

Audisport9, 2017-05-18:
Any idea when BLE will be fully supported?

WALLTECH, 2017-05-18:
Same, it's one of the core features of this module and it's been months

Audisport9, 2017-05-18:
For Arduino? I haven't found any arduino documentation and/or examples except simple ble and wifi/ble switch.  If there is can you point me in to it? Thanks

jestebanes, 2017-05-18:
I'm also waiting, and now even the simple ble example is not working!

mxwang101, 2017-05-18:
Yes，We do need this. And it is why use esp32.
On 05/19/2017 05:30, Audisport9 wrote:

For Arduino? I haven't found any arduino documentation and/or examples except simple ble and wifi/ble switch. If there is can you point me in to it? Thanks

jestebanes, 2017-05-19:
As far as I hack, the programming of BLE in Arduino is based on bluedroid stack, that is the same you can find in esp-idf. For this reason, the gat_server or gatt_client examples on esp-idf are easily translated to Arduino.
But by now the Ble device lost the capacity of advertising and for this reason, I can't test any of these examples to confirm my theory.

me-no-dev, 2017-05-19:
GUYS... BLE is supported... you have access to all functions that IDF has. As you have already seen even the simples of the simple codes is still too volatile in IDF (SimpleBLE not working), so how do you suppose me to know where things will go and also to always keep a close eye to every change in IDF not to break more stuff. I have sed this many times now. Once BT stack is stable and less frequently updated/changed, I will sit down and figure out to expose another more comfortable API for you. Until then, use IDFs API (there are examples also).

me-no-dev, 2017-05-19:
I did a good couple of hours digging through this. And there is no apparent reason why it would not work, but it does not. Callbacks are not called and the code can not progress. The included example in IDF has not changed either, so there are no clues to what has changed or what can be done. Maybe good time to remove that SimpleBLE example and be done for now.
My test GATT service is working, so not all is broken.

Audisport9, 2017-05-22:
@me-no-dev , Could you please post your test GATT service as an example please?

me-no-dev, 2017-05-22:
it's not a working example. Why would you want that :D

me-no-dev, 2017-05-22:
BTW the SimpleBLE is now working

Audisport9, 2017-05-22:
@me-no-dev ... If your GATT service isn't working why would you post that it is: 

me-no-dev > I did a good couple of hours digging through this. And there is no apparent reason why it would not work, but it does not. Callbacks are not called and the code can not progress. The included example in IDF has not changed either, so there are no clues to what has changed or what can be done. Maybe good time to remove that SimpleBLE example and be done for now.
**My test GATT service is working, so not all is broken.**

This thread is for arduino IDE users. So, if ESP32 is supporting arduino IDE, then Espressif needs to have a working Arduino IDE library for BLE, etc. 

My question to you, is are you writing your code in Arduino IDE or something else? Because if it is Arduino, then the Simple BLE example doesn't work. Sorry!! The ESP32 does not advertise. This is the same with all of the BLE examples for ESP32. None of them advertise the device.

me-no-dev, 2017-05-22:
I compiled today against todays Arduino code and it works. When I said that my GATT server is working, I meant that it advertises itself and so on, not that it's intended functionality is working properly. I did commit a change to SimpleBLE today and that is why I am saying that it's now working (compiled in Arduino IDE). If you are running under older IDF or something else, I can not help, but with today's Arduino and IDF it does.
I develop in Sloeber (Eclipse+Arduino) and write all code in Arduino style, not IDF.

me-no-dev, 2017-05-22:
https://github.com/espressif/arduino-esp32/commit/db09ca8c162a096271333ff3fbdfea1bf5be761c

Audisport9, 2017-05-22:
@me-no-dev ... thanks... yes BLE is now advertising...

What i was getting at in my original post is..... When will Espressif get a "FULL WORKING" library with api for BLE??? Simple BLE is the only BLE library and just now only advertises, that is it.. No other functionality or capability. So, what good is the ESP32 for arduino BLE users if BLE only advertises?

Audisport9, 2017-05-22:
The Arduino support for BLE on the ESP32 is like buying a car advertised with a super badass stereo that only turns the screen ON, NO sound.... Pretty useless :(

me-no-dev, 2017-05-22:
You have BT API , You have BT Hardware. What else do you need? You have all that you are complaining not to have. Your comments are what is pretty useless

closed 2017-05-22 by me-no-dev

andrei-ivanov, 2017-05-23:
Maybe pretty useless to you, but I agree with the whining.
When I bought the board in November, I did it for the same reason, WiFi and BLE, just to discover that BLE is not really there. I have patience, but quite some time has passed since then and there isn't even a plan for the Arduino support.
And since I'm not a C programmer, the Arduino part is very important for me and it seems a lot others, so just the IDF is not helping us :(
I understand that IDF is still a moving target, but I hope you'll still consider an implementation.

me-no-dev, 2017-05-23:
Of course I will... But until things have settled and API is clear, I will not release abstractions that will be hard to support. Hardware features and software features are different things. As I said MANY times already, BT is the part that is most worked on in IDF. When this is done and APIs can be properly tested, this is pointless. I don't think that any of you commenting even have a clue of the inner works of the BT protocols, or you will not be putting such pressure for some imaginary simpler API.
And last but not least: THIS IS AN OPEN SOURCE PROJECT! You can contribute code to it! I can not write everything and I do have lots of things lined up (BT one of them).

andrei-ivanov, 2017-05-23:
Indeed, I really don't have a clue about how BT works and, more important, how far is IDF from finishing the work on it. Is there any public roadmap for IDF?
I have some basic C++ knowledge, hence an Arduino user and unable to contribute to the implementation.
Regarding abstractions, I was hoping that Arduino would already have them already, but I come from the Java world where we have API specs for most of the libraries.

Sorry for being annoying :-D

me-no-dev, 2017-05-23:
AFAIK there already is an API implemented for one of the Arduinos, but is still no help until I have profound knowledge of the available API through IDF and how it all works. I have already gained quite a bit, but there are still big holes that either do not exist as code, or I have not found the correct way to use it (pairing for example). Then all have to be written in a way that is not getting Arduino users into trouble. Writing Arduino libs is not like writing an application. Libs might get used and abused in many ways by the users and application is a final product.

AdySan, 2017-05-23:
Some constructive feedback: perhaps people are looking for a library like [BLEPeripheral](https://github.com/sandeepmistry/arduino-BLEPeripheral)? The examples in there should satisfy most people. I have no idea how platform agnostic it is. Only used it with the [nRF5](https://github.com/sandeepmistry/arduino-nRF5) arduino core

mxwang101, 2017-05-24:
We do need it!
On 05/23/2017 21:48, Aditya Tannu wrote:

Some constructive feedback: perhaps people are looking for a library like BLEPeripheral? The examples in there should satisfy most people. I have no idea how platform agnostic it is. Only used it with the nRF5 arduino core

VikOlliver, 2017-07-13:
I for one appreciate the work me-no-dev is putting in here. They're aiming at a moving target. In the dark. I'm still trying to figure out how HOGP and BT Classic HID work, which makes me more sympathetic to the poor buggers working on the next level down.

lucascoelhof, 2017-07-26:
Hi folks,

I modified the SimpleBLE libraries a bit and I managed to advertise some data using the "Manufacturer Specific Data" field. It's only 8 bytes though, but I promise to work on this code on the following days to use the other fields too. Here is my repo: https://github.com/lucascoelhof/ESP32BleAdvertise

After some tests we can make a pull request to include this code to SimpleBLE.

VikOlliver, 2017-07-26:
Baby steps :) Thanks. Will try it out when I get back to base.