espressif/arduino-esp32 #384: micros() and millis() are uint32_t (unsigned int), instead of unsigned long

erenfro, 2017-05-18:
Arduino-esp32/cores/esp32/esp32-hal.h

Shows that millis() and micros() are uint32_t types, which is unsigned int's. This is incompatible with the ESP8266 code which was unsigned long types. This causes portability to go downhill due to type differences.

me-no-dev, 2017-05-19:
unsigned long on AVR is 32 bit ;) Just FYI.
I tried making the functions 64 bit at some point, but ran into issues, I'll give it another go... maybe use different method for millis();

erenfro, 2017-05-19:
I'm not sure if it's a bit issue, but it's definitely a typecast issue in respects to keeping with the Arduino SDK standard at the least. 

Like I said, I primarily noticed this when porting ESP8266 code to ESP32 and one library I use to measure pulse width frequencies failed to work because millis() and micros() were not the same typecast as was expected, which was unsigned long. When using millis and micros, I've always seen unsigned long's used in reference to it.

kriswiner, 2017-05-20:
unsigned long in ambiguous. Why not use uint64_t, for example, this is specific and cannot be confused with anything else.

igrr, 2017-05-20:
As @erenfro correctly noted, this isn't a bit width issue, as both int and long are 32 bit on Xtensa. However in C++ `int` and `long` are different types, even if they are of the same size. So the only way to allow libraries to be portable between ESP32 Arduino core and other cores is to use `unsigned long` for the return value. Yes, this is ambiguous (there could be a platform with 64-bit long), but language-wise this is the only way to remain compatible with other cores.

me-no-dev, 2017-05-22:
Merged :)

closed 2017-05-22 by me-no-dev