espressif/arduino-esp32 #385: SD library error

clowrey, 2017-05-19:
Upon compiling a fairly large program with:
#include "FS.h"
#include "SD.h"
#include "SPI.h"

inside setup:
 SD.begin();

I get these errors:

```
\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib\libfatfs.a(diskio.o): In function `get_fattime':

/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/fatfs/src/diskio.c:82:(.text.get_fattime+0x6): dangerous relocation: windowed longcall crosses 1GB boundary; return may fail: time

collect2.exe: error: ld returned 1 exit status
```
Any clue on the cause here? Do you think it is a direct problem with the library or am I hitting some hard size limits? 

My sketch before including the SD library only uses up this much space:
```
Sketch uses 134698 bytes (12%) of program storage space. Maximum is 1044464 bytes.
Global variables use 15572 bytes (5%) of dynamic memory, leaving 279340 bytes for local variables. Maximum is 294912 bytes.
```

me-no-dev, 2017-05-19:
I do not think that this is a sketch related error ;) I wonder what is the reason for that... @igrr any clues?

clowrey, 2017-05-19:
Okay this was not a problem with the SD library after all.

Real issue was a miss-used variable named "time" in one of the other source files of this project (that I did not write ;)... It compiled fine without the FS.h included. But FS.h tried to use the variable "time" in get_fattime(void) this caused the linker error.. 

```
DWORD get_fattime(void)
{
    time_t t = time(NULL);
```

closed 2017-05-19 by clowrey