espressif/arduino-esp32 #386: Didn't work on Windows XP

adxelec, 2017-05-19:
I installed this into Arduino 1.8.2 on XPSP3 and it failed to open the serial port. That's because the latest pyserial is not supported, being a newb I was unable to get esptools.exe to build properly with py2exe, so installed python 2.7, added (with easy_install) latest pyserial (pyserial-3.0.1.win32.exe) that claims to work under XP, then patched
C:\arduino-1.8.2\hardware\espressif\esp32\platform.txt
(directory structure is from portable install needed for ESP8266)
with this line:

`tools.esptool.cmd.windows=C:\Python27\python.exe "{runtime.platform.path}/tools/esptool.py"`

That worked and the example uploaded and ran fine first time. Everything else worked fine under XP so I thought it worthy posting here.

me-no-dev, 2017-05-19:
oh wow :) Windows XP FTW! Damn that 16 years old OS

ESAlexhi, 2017-05-20:
I compiled esptool  and it works in  XPSP3

adxelec, 2017-05-20:
Oops, I didn't think to look how to do that, it's here:
https://www.esp32.com/viewtopic.php?t=1029

It seems py2exe was never meant to work. Using pyinstaller worked first time. After moving the resultant esptool.exe to replace the old one and de-patching the line in my earlier post, it uploaded to the ESP32 first time.

Now, to get this working on a Win 95 machine I have in my hardware area... (via RDP remote serial, with the UXGA monitor I sometimes forget it's running 95!)

me-no-dev, 2017-05-21:
hahahaha! Yeah, I use pyinstaller to create the binaries ;)

closed 2018-04-19 by everslick