espressif/arduino-esp32 #387: ESPmDNS Seems not to work.

Kraay89, 2017-05-19:
_Not sure if this is even close to allowed here (my first ever issue created on GH)._

After adding the following to my program in the appropriate place in a slightly altered wifi server example (after WiFi.begin... and before server.begin)

`WiFi.setHostname("MyDevice");`

My program compiles and everything works as advertised
I can't reach the device from a browser as i was expecting to... Is this correct behavior and did i misinterpret how ESPmDNS works and what it provides or am i missing something?

erenfro, 2017-05-19:
I'm having this same issue. In my case, I'm using the ArduinoOTA which does ultimately set the mDNS name itself, and it too started failing since a recent git pull. It worked at one point prior, but stopped. I just wish I recalled which checkout tag I was on at the time I noticed it working.

copercini, 2017-09-09:
@Kraay89 try with all letters in lowercase ;)

zsiroslaci, 2018-01-31:
I just tried ESPmDNS example. I was compiled without any error, but it not work.
I suppose that MDNS.begin("esp32"); should include IPAddress as well , or something to bond to it, but no exaple what and how. In case of  Esp8266, the WiFi.localIP() is included an that is working. in ESP32 it does not work, giving compiling error. What is the solution?

baggior, 2018-02-03:
I can confirm that mdns is not working to me too
no error shown but the service is not dicovered.
board = esp32dev

(The same code on ESP8266 works)

beegee-tokyo, 2018-02-05:
As commented [here](https://github.com/espressif/arduino-esp32/issues/962) there are differences between the ESP8266 version of the library and the ESP32 version
- Service name must be all lower case and start with an '_'
- Protocol must be all lower case and start with an '_'
If you call the same way as in ESP8266
```cpp
MDNS.addService("osc", "udp", 4500);
MDNS.addServiceTxt("osc", "udp", "board", "ESP8266");
```
it will fail (without error) on ESP32. You need to call with
```cpp
MDNS.addService("_osc", "_udp", 4500);
MDNS.addServiceTxt("_osc", "_udp", "board", "ESP32");
```

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

MkLHX, 2018-12-17:
Hi, i have the same problem now on ESP32 with ESPmDNS lib.

I understand the reason why this issue is closed but where put the searching result about fix it.

I tried all i can seen here. But nothing help me.

I tried the mDNS_Web_Server example and just change "http" and "tcp" to "_http" and "_tcp" on
 ```    // Add service to MDNS-SD
    MDNS.addService("_http", "_tcp", 80);```

But when i try to acces on http://esp32.local
nothing happens

I'm using ESP32 Dev Module board / PlatformIO / Espressif Arduino-Esp

elchupokabrapolisetty, 2020-01-03:
@MkLHX 

Add 
```
MDNS.addService("_http", "_tcp", 80);
MDNS.addServiceTxt("_http", "_tcp", "board", "ESP32");
```

make sure that both the statements are added.
I had the same problem and it was fixed after i added the Second line