espressif/arduino-esp32 #390: No touch wake up?

tokschi, 2017-05-19:
I tried this to perform touch wake up:

`// ESP32 Touch Test
#include "esp_deep_sleep.h"

int k=0;
uint64_t duration= 5000000;

void setup() {
  Serial.begin(115200);
  delay(1000); // give me time to bring up serial monitor
  Serial.println("\n ESP32 Touch Test\n");
  
  touch_pad_init();
  touch_pad_t touch_ret= esp_deep_sleep_get_touchpad_wakeup_status();
  Serial.println(touch_ret);
  Serial.println("");
}

void loop() {

  // Touch0, T0, is on GPIO4
  // read Touch to test functionality
  uint16_t val;
  touch_pad_read(TOUCH_PAD_NUM0, &val);
  Serial.println(val);
  delay(1000);

  // after ten iterations goto sleep
  k++;
  if (k>10) {
    touch_pad_config(TOUCH_PAD_NUM0, 1000);
    esp_deep_sleep_enable_touchpad_wakeup();
    esp_deep_sleep_start();
  }
}
`

I know, this isn't the arduino style, but reading the values works fine. So I thought wakeup should work, too. Is it intended, that I can use this libs?

Or do we need this fix? [https://github.com/espressif/esp-idf/issues/542]

tokschi, 2017-05-19:
Sorry for the format errors. Will try the preview next time.

igrr, 2017-05-20:
What range of touch values do you get before deep sleep (with/without touching the pad?).

Could you also try some other pin, such as TOUCH7 (GPIO27)?

tokschi, 2017-05-20:
Thanks for your fast reply. Yes I played with the threshhold. I had values between 500 and 1300 w/o touching and lower than 100 with touching. Then started with threshhold 300 and increased it to 1000. So 1000 should trigger wakeup by accident rather immidiately.

But wakeup by timer workes fine with the same library.

I'll try the other port this evening. What was your intention for suggesting the other port?

tokschi, 2017-05-20:
Touch7 does'n solve the problem, unfortunately. For me it sounds like this prroblem: https://github.com/espressif/esp-idf/issues/542

But I didn't find the sources they fixed in the arduino environment.

me-no-dev, 2017-05-21:
@tokschi yjay fix seems merged in IDF, therefore it's already implemented here as well.

copercini, 2017-08-18:
Implemented here: https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/DeepSleep/TouchWakeUp/TouchWakeUp.ino

closed 2017-08-18 by copercini

beats0126, 2018-06-28:
@copercini  may i ask a question?
i using mpr121 touch sensor to wake up my esp32, i connect the irq pin fron mpr121 to esp32 pin 15, and in the code i set it to gpio_num_03. the problem i have is the esp32 keep wakeup. it is any suggestion to use or do touch sensor to wakeup esp32?