espressif/arduino-esp32 #391: WiFi not connecting with latest git version

retsifp, 2017-05-21:
Is it just me? With the latest git version (51a4432ca8e71be202358ceb068f3047bb8ad762), Wifi is only connecting when deleting the flash before flashing the sketch. When flashing the same sketch again, the ESP32 hangs in the connecting loop (keeps outputting dots).

Here is my sketch (pretty much the example):

```
void setup() {
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}
```

When downgrading to a older version (via `git reset --hard 4b47402afdb198cd4a62f0356dbcb20ab0bc3845`), everything works fine.
I assume it's about this commit: 4765554afd97a4ba6da55a45ecc324e92a514246, but I don't really know.

me-no-dev, 2017-05-21:
I'm investigating some WiFi issues as well :) might not be just you

lrmoreno007, 2017-05-21:
Try with this:
```
void setup() {
    Serial.begin(115200);
    WiFi.enableSTA(true);
    delay(2000);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}
```

retsifp, 2017-05-21:
Thanks, works! 😄

me-no-dev, 2017-05-22:
I just committed some changes. Can I bother you to try the old code?

retsifp, 2017-05-22:
Thanks, works! 😄

retsifp, 2017-05-22:
Whoop, that was too fast. 😉  Worked only the first time after pulling, but didn't after the first re-flash.

me-no-dev, 2017-05-22:
I reflashed and restarted a few times. Connecting fine every time (with the sketch in the OP)

retsifp, 2017-05-22:
Did you try reflashing the same sketch? Does work for me for the first time uploading a sketch after deleting the flash with esptool, but stops working after the first re-flash.

me-no-dev, 2017-05-22:
I reflashed at lest 10 times

lrmoreno007, 2017-05-23:
I tried it again and:

In fixed IP mode:
With WiFi.enableSTA(true) work always OK and ping are OK,
Without WiFi.enableSTA(true) work OK first time always after erase_flash, and the system always return WL_CONNECTED = True, but if I ping the IP from any device to ESP32 IP sometimes "Request timed out" or "destination host unreachable".
---> In fixed mode, **if I activate Debug Mode everything work OK** and it's the feedback.
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 7 - STA_GOT_IP

In DHCP mode:
With WiFi.enableSTA(true) work always OK and ping are Ok.
Without WiFi.enableSTA(true) work OK first time always after erase_flash, sometimes after upload sketch and randomly after reset. When failure occurs WL_CONNECTED never True. 
---> In DHCP Mode and when failure occurs, Debug mode does not return any information and don't make any difference.

I don't feel any difference between soft reset (ESP.restart()) and hard reset (button reset or reset in serial software)

My hardware and software:
ESP32-ST
80 Mhz
Windows 10
Git version db09ca8
Arduino IDE and / or Eclipse Neon with sloeber

me-no-dev, 2017-05-23:
Found it!

me-no-dev, 2017-05-23:
Please retest, but it should/will be ok

closed 2017-05-23 by me-no-dev

lrmoreno007, 2017-05-23:
Hi, with DHCP problem is solved (connection and ping OK). 

With fixed IP the problem is solved (connection and ping OK) but I must use this order:
WiFi.begin(blablabla)
WiFi.config(blablabla)

Before I was doing it just the other way around and this work in ESP8266, but it's no problem for me. 
WiFi.config(blablabla)
WiFi.begin(blablabla)

Thanks!!!

Dominik-H, 2017-06-04:
Just today I've got the same problem on DHCP. I resolved it temporarily with enableSTA. I did not test it with fixed IP.

bioyavuz, 2018-01-14:
I faced the same problem and tried to understand the reason. Updating the latest idf-core etc. made no difference. 

On every newly started sketch everything seemed fine. But copying the same code into the sketch on which I am working on, was failing again.

So I decided to find out the difference between these two. I spent my last few days on this issue.  Finally, my problem disappeared with an interesting finding:

Whenever I  use any "Core Debug Level" other than "None" problem persisted. Changing Tools->Core Debug Level" to "None" solves the problem disappears.  No connection problem at al.

Can someone explain why.