espressif/arduino-esp32 #392: pins_arduino.h: No such file or directory

joeybab3, 2017-05-21:
No matter which board or example I use, I keep getting the error:
`C:\Users\meusername\Documents\Arduino\hardware\espressif\esp32\cores\esp32/Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory`

Even BareMinimum.ino from the arduino examples will not compile.

me-no-dev, 2017-05-22:
something is wrong with your installation. the "missing" file is in the variant folder. Which board did you select?

joeybab3, 2017-05-23:
I tried esp32 dev board, node32, nodemcu-32s... I tried installing on another computer and it works however so it does appear my issue is a one off...

closed 2017-05-23 by joeybab3

shepting, 2019-04-03:
I've run into this as well for the record: 
```
In file included from sketch/Blink.ino.cpp:1:0:
/Users/steven_hepting/Library/Arduino15/packages/esp32/hardware/esp32/1.0.1/cores/esp32/Arduino.h:190:26: fatal error: pins_arduino.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board SparkFun LoRa Gateway 1-Channel.
```

beegee-tokyo, 2019-04-04:
I have the same board and it compiles just fine.
Did you follow all steps from Install the [LoRa Gateway 1-Channel Board Definition](https://learn.sparkfun.com/tutorials/sparkfun-lora-gateway-1-channel-hookup-guide/programming-the-esp32-with-arduino)? It sounds to me as if the "variant" is missing.