espressif/arduino-esp32 #393: CPU exception on write to char*

hsgentry, 2017-05-21:
I inherited some code that compiled and ran fine on ESP8266 with the IDE 1.8.2.  When I tried to run it on the ESP32 it compiled fine but  threw an exception.

 char* x="1234567";
  
Any attempt to write to x results in a CPU exception:  Guru Meditation Error of type LoadStoreError occurred on Core 1. Exception was not handled.

If I had to guess I would bet that the char array was being allocated in some memory that can be read but not written.

I would not have coded it this way but I inherited the code.  It was easy to recode to work around. 

I tested:

char x[]="12345" ;

and it can be written to.

projectgus, 2017-05-22:
This happened to work on ESP8266, but it's crashing "correctly" on ESP32 unfortunately. Your guess that the string is being allocated in read-only memory is correct, but it's also intended behaviour.

Allocating a string literal like "12345", gcc will put it in the read-only data (.rodata) section. This section is read-only on most platforms, but ESP8266 happened to store .rodata in RAM so it was actually writeable. This causes a bunch of related problems on ESP8266, because all constants (log messages, other data, etc.) take up RAM. However it has the unusual side effect that constants are actually editable.

Even though `char *x= "12345";` lets you create a non-const pointer, it still points to const .rodata. I'm not certain why the compiler can't/doesn't issue a warning on this behaviour. 

The fix you found of declaring x as a mutable char array is a good fix, and will work on both ESP8266 & ESP32.

hsgentry, 2017-05-22:
I suspected that and I also expected that it was deliberate, but thought
that since the behavior was different and that there was no compiler error
it was worth reporting.  I wrote C, C++ and C# code for almost 40 years and
I don't ever remember doing it that way.  As I said I inherited that code
and frankly did not look closely at it since it worked on the 8266.  I only
looked at it when it failed on the 32.

Thanks for the information.

Sam

igrr, 2017-05-22:
> I'm not certain why the compiler can't/doesn't issue a warning on this behaviour, but it doesn't.

By default, Arduino compiles without `-Wall`.
https://github.com/espressif/arduino-esp32/blob/master/platform.txt#L18-L20

If you select "All" warnings in Arduino > Preferences, you will get a warning similar to this:

```
/var/folders/yh/gpd_gkgs0nn_h9rhfqrgwbcc0000gp/T/arduino_modified_sketch_51601/sketch_may22a.ino:3:16: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
   char * x = "12345";
              ^
```

closed 2018-04-19 by everslick