espressif/arduino-esp32 #395: WiFiServer partly working

bbx10, 2017-05-23:
@me-no-dev I have WiFiServer partly working. The following examples fail: FSBrowser, SDWebServer, Updater. Should I submit a PR? I will not have much time to work on this for a week or two.

me-no-dev, 2017-05-23:
Not for the WebServer ;) That needs to wait a bit. I'm looking into other classes to import (DNS, HTTPClient).

bbx10, 2017-05-26:
Ports of ESP8266WebServer and DNSServer are on my [fork](https://github.com/bbx10/arduino-esp32/tree/add_webserver_and_dnsserver). See the branch "add_webserver_and_dnsserver".

The DNSServer examples depend on WebServer so WebServer must be added at the same time.

Three WebServer examples are not included because they depend on other libraries. I do not think this indicates a problem with the WebServer library. FSBrowser depends on SPIFFS which is not available. I do not understand enough about WebUpdate to port it. SDWebServer depends on SD card which is not working for me. Perhaps someone else would like to take a shot at making these work.

me-no-dev, 2017-05-26:
We discussed this with @igrr a bit and have decided to open separate repos for libs like the web server, dns and such, so we do not need to support two separate copies of the same libs. Base classes will stay in their respective cores, but libs should go their own way. That way tracking and contributing fixes will be much easier. There is a base FS class in ESP32 as well, so FS code can already be made to work between both chips. The rest should be a breeze as well.

bbx10, 2017-05-26:
@me-no-dev @igrr Great idea! The library and example differences between ESP8266 and ESP32 can easily be handled by #ifdef's.

There are some changes to the WiFiClient and WiFiServer classes for compatibility with the libraries. Let me know if you want a PR for those changes.

roboticboyer, 2017-05-26:
Please link the new repos of the libraries
All the community will appreciate ;)

bbx10, 2017-05-30:
@igrr @me-no-dev PR #404 has useful additions for WebServer and DNSServer libraries.

Once the PR has been merged, I plan to create unofficial repos for WebServer and DNSServer unless you think I should wait for the official repos. The libraries are working well enough for me to get some of my projects ported to ESP32. The addition of HTTPClient has also helped a lot.

I started with the latest ESP8266WebServer/DNSServer libraries from esp8266/Arduino then added ESP32 changes with a generous dose of #ifdef. In theory, the libraries should work on ESP8266 and ESP32. In practice, minimal testing has been done on ESP32 and none on ESP8266.

me-no-dev, 2017-05-30:
Please do not spin off "unofficial" repos, so we do not get users into issues later ;)

bbx10, 2017-05-30:
OK, I will wait for the official repos.

salqadri, 2017-06-10:
Just saw this issue. I had created a PR (#430) reacting to a more recently-filed issue, but it looks like my PR is similar to the PR for this issue. +1 to the idea of creating common lib repos for both ESP devices.

schweini, 2018-05-01:
Has there been any movement on this issue of merging @bbx10 's WebServer and DnsServer fixes? 

It seems like a small but very convenient change?

copercini, 2018-06-27:
Please update to last code, it's now supporting webserver =)
https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer

closed 2018-06-27 by copercini